robotology/wb-toolbox #180: Add OSQP block to deprecated QpOases  block

traversaro, 2019-08-23:
#### Summary

It would be great to add a block for solving Quadratic Programming (QP) optimization problems using [OSQP](https://osqp.readthedocs.io/en/latest/index.html) (and optionally https://github.com/robotology/osqp-eigen) instead of the existing block that uses [qpOASES](https://projects.coin-or.org/qpOASES). 

#### Motivation
See  on why it would be great to avoid the qpOASES dependency. 

#### Additional context

It may be even better to wrap in WB-Toolbox some generic interface with pluggable backend for QP programming, such as the one provided in iDynTree, Drake or RobOptim. However, this would be a bit more complex, and in the iDynTree case would also involve some work on the iDynTree side, while simple creating an OSQP block would be trivial, especially given that it already exists a

traversaro, 2019-08-23:
Related comment from @diegoferigo : https://github.com/robotology-dependencies/qpOASES/issues/4#issuecomment-524298966 .

traversaro, 2019-11-20:
Code of the existing block: 
* https://github.com/robotology/wb-toolbox/blob/f110c4b641a09a4bb392161db2ab79d0afc0adbe/toolbox/library/include/WBToolbox/Block/QpOases.h
* https://github.com/robotology/wb-toolbox/blob/f110c4b641a09a4bb392161db2ab79d0afc0adbe/toolbox/library/src/QpOases.cpp

DanielePucci, 2019-11-20:
CC
@gabrielenava @S-Dafarra

gabrielenava, 2020-03-06:
cc @CarlottaSartore

nunoguedelha, 2021-02-22:
Hi @diegoferigo , could you assign this to me please, I can't do it myself.

traversaro, 2021-02-22:
Done @nunoguedelha !

traversaro, 2021-05-18:
@nunoguedelha I saw that the issue was in Ice-Box, I implemented a first version of it this morning.

traversaro, 2021-05-18:
@diegoferigo `export_libraries` works nice also on Windows, cool:
[image]

closed 2021-05-24 by traversaro