robotology/wb-toolbox #186: Unable to connect to multiple remotecontrolboard clients

lrapetti, 2020-01-30:
#### Description

I am trying to run a Simulink project using WBToolbox to control two robots in a Gazebo simulation. The project was working a couple of months ago (in the middle for sure I updated `YARP` to more recent versions, and eventually other folders, not sure of all the changes) but now I am no longer able to have it running. Commenting out one or the other robot the project is working, but when I try to run it with both the robots, it is failing giving following errors:
```
...
yarp: No connection to nameserver
yarp: *** try running: yarp detect ***
yarp: No connection to nameserver
yarp: *** try running: yarp detect ***
yarp: Receiving input from /iCub2/right_leg/stateExt:o to /WBT/torqueControlBalancingRobot2Configuration/iCub2/right_leg/stateExt:i using udp
yarp: Removing input from /iCub2/right_leg/stateExt:o to /WBT/torqueControlBalancingRobot2Configuration/iCub2/right_leg/stateExt:i
yarpdev: ***ERROR*** driver <remote_controlboard> was found but could not open
yarp: Removing output from 
...
```
(full terminal and Simulink diagnostic output can be found below)

The problem is somehow similar to https://github.com/robotology/yarp/issues/1850.

#### Additional context

**Simulink configuration**
Outer view of the Simulink project:
[image]
Each robot block contains the `Config` and the WBtoolbox boxes as in the following picture
[image]

For each robot I have a different configuration:
- Robot 1
```
WBTConfigRobot_1 = 

  Configuration with properties:

             RobotName: 'iCub1'
              UrdfFile: 'model.urdf'
      ControlledJoints: {1×23 cell}
    ControlBoardsNames: {'torso'  'left_arm'  'right_arm'  'left_leg'  'right_leg'}
             LocalName: 'WBT'
         GravityVector: [0 0 -9.8100]
    ValidConfiguration: 1
```
- Robot 2
```
WBTConfigRobot_2 = 

  Configuration with properties:

             RobotName: 'iCub2'
              UrdfFile: 'model.urdf'
      ControlledJoints: {1×23 cell}
    ControlBoardsNames: {'torso'  'left_arm'  'right_arm'  'left_leg'  'right_leg'}
             LocalName: 'WBT'
         GravityVector: [0 0 -9.8100]
    ValidConfiguration: 1
```

**terminals outputs**
Matlab terminal:
```
yarp: Port /WBT/torqueControlBalancingRobot1Configuration/iCub1/torso/rpc:o active at tcp://10.240.2.21:10086/
yarp: Port /WBT/torqueControlBalancingRobot1Configuration/iCub1/torso/command:o active at tcp://10.240.2.21:10087/
yarp: Port /WBT/torqueControlBalancingRobot1Configuration/iCub1/torso/stateExt:i active at tcp://10.240.2.21:10088/
yarp: Sending output from /WBT/torqueControlBalancingRobot1Configuration/iCub1/torso/rpc:o to /iCub1/torso/rpc:i using tcp
yarp: Sending output from /WBT/torqueControlBalancingRobot1Configuration/iCub1/torso/command:o to /iCub1/torso/command:i using udp
yarp: Receiving input from /iCub1/torso/stateExt:o to /WBT/torqueControlBalancingRobot1Configuration/iCub1/torso/stateExt:i using udp
yarp: Port /WBT/torqueControlBalancingRobot1Configuration/iCub1/left_arm/rpc:o active at tcp://10.240.2.21:10089/
yarp: Port /WBT/torqueControlBalancingRobot1Configuration/iCub1/left_arm/command:o active at tcp://10.240.2.21:10090/
yarp: Port /WBT/torqueControlBalancingRobot1Configuration/iCub1/left_arm/stateExt:i active at tcp://10.240.2.21:10091/
yarp: Sending output from /WBT/torqueControlBalancingRobot1Configuration/iCub1/left_arm/rpc:o to /iCub1/left_arm/rpc:i using tcp
[... 231 lines cut ...]
registration name /iCub2/right_arm/state:o ip 10.240.2.21 port 10076 type tcp
registration name /iCub2/right_arm/stateExt:o ip 10.240.2.21 port 10077 type tcp
registration name /iCub2/right_hand/inertialMTB ip 10.240.2.21 port 10059 type tcp
registration name /iCub2/right_hand/inertialMTB/rpc:i ip 10.240.2.21 port 10058 type tcp
registration name /iCub2/right_leg/command:i ip 10.240.2.21 port 10079 type tcp
registration name /iCub2/right_leg/inertialMTB ip 10.240.2.21 port 10055 type tcp
registration name /iCub2/right_leg/inertialMTB/rpc:i ip 10.240.2.21 port 10054 type tcp
registration name /iCub2/right_leg/rpc:i ip 10.240.2.21 port 10078 type tcp
registration name /iCub2/right_leg/state:o ip 10.240.2.21 port 10080 type tcp
registration name /iCub2/right_leg/stateExt:o ip 10.240.2.21 port 10081 type tcp
registration name /iCub2/torso/command:i ip 10.240.2.21 port 10063 type tcp
registration name /iCub2/torso/inertialMTB ip 10.240.2.21 port 10061 type tcp
registration name /iCub2/torso/inertialMTB/rpc:i ip 10.240.2.21 port 10060 type tcp
registration name /iCub2/torso/rpc:i ip 10.240.2.21 port 10062 type tcp
registration name /iCub2/torso/state:o ip 10.240.2.21 port 10064 type tcp
registration name /iCub2/torso/stateExt:o ip 10.240.2.21 port 10065 type tcp
...
```

#### Additional information

- OS: `MacOS 10.14.6`
- YARP version: `3.3.1`
-  WBtoolbox: `master` (114ac0e76068399c51150cb55c5922634d199ad3)
- Matlab Version: `2019b`

lrapetti, 2020-01-30:
I tried also with a different project that connects to multiple robots and I am getting the same error.

lrapetti, 2020-01-30:
cc @Yeshasvitvs @diegoferigo

traversaro, 2020-01-30:
I think the original intuition by @barbalberto in the comment in https://github.com/robotology/yarp/issues/1850#issuecomment-421978615 may be a good hint. 

Nevertheless, to have something more actionable, I think you should proceed as in the following. As nothing changed  recently in WB-Toolbox or BlockFactory, indeed a change of YARP can be a good candidate for creating this issue. First of all, find a YARP commit that is working fine, based on your original comment I would try to see if YARP v3.2.2 is working  fine instead. If that tag is working fine while v3.3.1 is not working, then we are  on a good track. Between v3.2.2 and  v3.3.1 there are ~280 commits (see https://github.com/robotology/yarp/compare/v3.2.2...v3.3.1), so by using [git bisect](https://git-scm.com/docs/git-bisect) it should just take ≈ log2(280) ≈ 9 tests to actually find the commit that introduced the problem.

traversaro, 2020-01-30:
For simplify the debugging, it would also be useful to have a Wireshark dump of the loopback traffic when you get this error.

lrapetti, 2020-01-31:
From a preliminary test I found out that is not working with YARP `3.2.2` but it is working with YARP `3.2.0`. I will try to investigate further about the exact commit.

traversaro, 2020-01-31:
> 
> 
> From a preliminary test I found out that is not working with YARP `3.2.2` but it is working with YARP `3.2.0`. I will try to investigate further about the exact commit.

If you are taking bets, I will put my money on https://github.com/robotology/yarp/commit/1c1d01b12b284e634ac8eb672202aa198bfae7ad or https://github.com/robotology/yarp/commit/6476dcf14acd3a87d72663be1dd31e162dbf472d .

lrapetti, 2020-01-31:
> From a preliminary test I found out that is not working with YARP `3.2.2` but it is working with YARP `3.2.0`. I will try to investigate further about the exact commit.

Unfortunately, after moving around with the commits I was no longer able to have it working (not even with `3.2.0`, but I am sure at a certain point it was working), which makes me a bit confused about the origin of the problem (or about my focus during those tests).

lrapetti, 2020-01-31:
> Unfortunately, after moving around with the commits I was no longer able to have it working (not even with `3.2.0`, but I am sure at a certain point it was working)

I discovered that if I compile `master` and then I switch to `3.2.0`, then it is working (on `3.2.0`). While moving trough intermediate commits and then to `3.2.0` apparently screwed the things.

traversaro, 2020-01-31:
> I discovered that if I compile `master` and then I switch to `3.2.0`, then it is working (on `3.2.0`). While moving trough intermediate commits and then to `3.2.0` apparently screwed the things.

Mhh, it could be something strange in the build system. It is probably a good idea to test each commit from a clean build.

traversaro, 2020-02-02:
Two things that I forgot: 
* To quickly compile even from a clean build, always use [`ccache`](https://ccache.dev/)
* It is also possible that the failure is not deterministic itself, so it is possible that it something that may go wrong also in 3.2.0, but in master has an higher probability of failure (for example if it is some kind of [race condition](https://en.wikipedia.org/wiki/Race_condition) or similar).

lrapetti, 2020-02-04:
> Mhh, it could be something strange in the build system. It is probably a good idea to test each commit from a clean build.

I did test with clean build of different commits and releases (recompiling `YARP`, `gazebo-yarp-plugins` and `wb-toolbox`) but I was not able to find a properly working commit (I moved from `3.2.0` on).

> * It is also possible that the failure is not deterministic itself, so it is possible that it something that may go wrong also in 3.2.0, but in master has an higher probability of failure (for example if it is some kind of [race condition](https://en.wikipedia.org/wiki/Race_condition) or similar).

Unfortunately this seems to me to be the most likely condition, I will report soon the output of some test remaining showing this condition.

lrapetti, 2020-02-04:
> Unfortunately this seems to me to be the most likely condition, I will report soon the output of some test remaining showing this condition.

The following results have been obtained, trough different runs in chronoligical order, with `YARP3.2.0` (where the coloumns indicate which software have been restarted after the previous try):

| result | Yarp server | Gz | Simulink | Matlab|
| :-: | :-: | :-: | :-: | :-: |
| **Fail** | x | x | x | x |
| **Fail** |  | x |  |  |
| **Success** | x | x | x | x |
| **Success** | x | x |  |  |
| **Success** | x | x |  |  |
| **Fail** | x | x | x | x |
| **Fail** |  | x |  |  |
| **Fail** |  | x |  |  |
| **Fail** |  | x |  |  |
| **Fail** | x | x |  |  |
| **Fail** | x | x |  |  |
| **Fail** | x | x | x |  |
| **Fail** | x | x | x |  |
| **Fail** | x | x | x | x |
| **Fail** | x | x | x | x |
| **Fail** | x | x | x | x |
| **Success** | x | x | x | x |
| **Success** | x | x | x |  |
| **Fail** (Gz crash) | x | x | x |  |
| **Success** | x | x | x | x |

lrapetti, 2020-02-04:
Similar with the previous comment, those are some test with `YARP3.3.101` (current master)

| result | Yarp server | Gz | Simulink | Matlab|
| :-: | :-: | :-: | :-: | :-: |
| **Fail** | x | x | x | x |
| **Fail** | x | x | x | x |
| **Fail** |  |  |  |  |
| **Fail** |  | x |  |  |
| **Fail** | x | x |  |  |
| **Success** | x | x | x | x |
| **Success** |  | x |  |  |
| **Success** | x | x |  |  |
| **Success** |  |  | x |  |
| **Success** |  |  | x | x |
| **Success** |  | x | x | x |
| **Success** | x | x | x | x |

So it doesn't seem to be much related to the YARP version

traversaro, 2020-02-04:
At this point, I think the problem is some yarpserver race condition that has been part of yarpserver for a long time, and for some reason (that could also be related to something on your PC) you are noticing it only now.

lrapetti, 2020-02-05:
I agree it is some kind of race condition, but would you exclude is something related to `wb-toolbox`?

I tried to debug a bit understanding where is is failing in [`initializeRemoteControlBoardRemapper`](https://github.com/robotology/wb-toolbox/blob/master/toolbox/base/src/RobotInterface.cpp#L146), and I have noticed something strange:
- the initialization is completed for the first model (`iCub1`)
- then it starts the initialization of the second model (`iCub2`)), and in the middle we see the errors and all the ports for the first model (`iCub1`) are removed
- then it starts again the initialization for the second model (`iCub2`), and this times it seems to be completed succesfully
- finally all the ports of the second model (`iCub2`) are removed

you can see this events in the following output (search for `START initializeRemoteControlBoardRemapper` and `FINISH initializeRemoteControlBoardRemapper`)
```
iCub1 : START initializeRemoteControlBoardRemapper 
yarp: Port /WBT/torqueControlBalancingModel1Robot1Configuration/iCub1/torso/rpc:o active at tcp://10.255.111.23:10087/
yarp: Port /WBT/torqueControlBalancingModel1Robot1Configuration/iCub1/torso/command:o active at tcp://10.255.111.23:10088/
yarp: Port /WBT/torqueControlBalancingModel1Robot1Configuration/iCub1/torso/stateExt:i active at tcp://10.255.111.23:10089/
yarp: Sending output from /WBT/torqueControlBalancingModel1Robot1Configuration/iCub1/torso/rpc:o to /iCub1/torso/rpc:i using tcp
yarp: Sending output from /WBT/torqueControlBalancingModel1Robot1Configuration/iCub1/torso/command:o to /iCub1/torso/command:i using udp
yarp: Receiving input from /iCub1/torso/stateExt:o to /WBT/torqueControlBalancingModel1Robot1Configuration/iCub1/torso/stateExt:i using udp
yarp: Port /WBT/torqueControlBalancingModel1Robot1Configuration/iCub1/left_arm/rpc:o active at tcp://10.255.111.23:10090/
yarp: Port /WBT/torqueControlBalancingModel1Robot1Configuration/iCub1/left_arm/command:o active at tcp://10.255.111.23:10091/
yarp: Port /WBT/torqueControlBalancingModel1Robot1Configuration/iCub1/left_arm/stateExt:i active at tcp://10.255.111.23:10092/
yarp: Sending output from /WBT/torqueControlBalancingModel1Robot1Configuration/iCub1/left_arm/rpc:o to /iCub1/left_arm/rpc:i using tcp
yarp: Sending output from /WBT/torqueControlBalancingModel1Robot1Configuration/iCub1/left_arm/command:o to /iCub1/left_arm/command:i using udp
yarp: Receiving input from /iCub1/left_arm/stateExt:o to /WBT/torqueControlBalancingModel1Robot1Configuration/iCub1/left_arm/stateExt:i using udp
yarp: Port /WBT/torqueControlBalancingModel1Robot1Configuration/iCub1/right_arm/rpc:o active at tcp://10.255.111.23:10093/
yarp: Port /WBT/torqueControlBalancingModel1Robot1Configuration/iCub1/right_arm/command:o active at tcp://10.255.111.23:10094/
yarp: Port /WBT/torqueControlBalancingModel1Robot1Configuration/iCub1/right_arm/stateExt:i active at tcp://10.255.111.23:10095/
yarp: Sending output from /WBT/torqueControlBalancingModel1Robot1Configuration/iCub1/right_arm/rpc:o to /iCub1/right_arm/rpc:i using tcp
yarp: Sending output from /WBT/torqueControlBalancingModel1Robot1Configuration/iCub1/right_arm/command:o to /iCub1/right_arm/command:i using udp
yarp: Receiving input from /iCub1/right_arm/stateExt:o to /WBT/torqueControlBalancingModel1Robot1Configuration/iCub1/right_arm/stateExt:i using udp
yarp: Port /WBT/torqueControlBalancingModel1Robot1Configuration/iCub1/left_leg/rpc:o active at tcp://10.255.111.23:10096/
yarp: Port /WBT/torqueControlBalancingModel1Robot1Configuration/iCub1/left_leg/command:o active at tcp://10.255.111.23:10097/
yarp: Port /WBT/torqueControlBalancingModel1Robot1Configuration/iCub1/left_leg/stateExt:i active at tcp://10.255.111.23:10098/
yarp: Sending output from /WBT/torqueControlBalancingModel1Robot1Configuration/iCub1/left_leg/rpc:o to /iCub1/left_leg/rpc:i using tcp
yarp: Sending output from /WBT/torqueControlBalancingModel1Robot1Configuration/iCub1/left_leg/command:o to /iCub1/left_leg/command:i using udp
yarp: Receiving input from /iCub1/left_leg/stateExt:o to /WBT/torqueControlBalancingModel1Robot1Configuration/iCub1/left_leg/stateExt:i using udp
yarp: Port /WBT/torqueControlBalancingModel1Robot1Configuration/iCub1/right_leg/rpc:o active at tcp://10.255.111.23:10099/
yarp: Port /WBT/torqueControlBalancingModel1Robot1Configuration/iCub1/right_leg/command:o active at tcp://10.255.111.23:10100/
yarp: Port /WBT/torqueControlBalancingModel1Robot1Configuration/iCub1/right_leg/stateExt:i active at tcp://10.255.111.23:10101/
yarp: Sending output from /WBT/torqueControlBalancingModel1Robot1Configuration/iCub1/right_leg/rpc:o to /iCub1/right_leg/rpc:i using tcp
yarp: Sending output from /WBT/torqueControlBalancingModel1Robot1Configuration/iCub1/right_leg/command:o to /iCub1/right_leg/command:i using udp
yarp: Receiving input from /iCub1/right_leg/stateExt:o to /WBT/torqueControlBalancingModel1Robot1Configuration/iCub1/right_leg/stateExt:i using udp
iCub1 : FINISH initializeRemoteControlBoardRemapper 
iCub2 : START initializeRemoteControlBoardRemapper 
yarp: Port /WBT/torqueControlBalancingModel2Robot2Configuration/iCub2/torso/rpc:o active at tcp://10.255.111.23:10102/
yarp: Port /WBT/torqueControlBalancingModel2Robot2Configuration/iCub2/torso/command:o active at tcp://10.255.111.23:10103/
yarp: Port /WBT/torqueControlBalancingModel2Robot2Configuration/iCub2/torso/stateExt:i active at tcp://10.255.111.23:10104/
[... cut ...]
yarp: Removing output from /WBT/torqueControlBalancingModel2Robot2Configuration/iCub2/right_leg/rpc:o to /iCub2/right_leg/rpc:i
yarp: Removing output from /WBT/torqueControlBalancingModel2Robot2Configuration/iCub2/right_leg/command:o to /iCub2/right_leg/command:i
yarp: Removing input from /iCub2/right_leg/stateExt:o to /WBT/torqueControlBalancingModel2Robot2Configuration/iCub2/right_leg/stateExt:i
```

aren't this sequence of events and the fact that the second model is initialized twice, strange?
@traversaro @diegoferigo

lrapetti, 2020-02-05:
Today I have tested the same code with the following configuration:
OS: `Ubuntu 18.04`
YARP: `3.2.101`
Matlab: `R2018a`

and I didn't experience the failure. Further investigation should be done in order to understand if the problem is related to Matlab version or OS.

lrapetti, 2020-02-06:
Adding some note on debugging the error, the error appears in the [`open()` call for the polyDriver](https://github.com/robotology/wb-toolbox/blob/master/toolbox/base/src/RobotInterface.cpp#L199). 
The passed arguments seems to be fine:
- iCub 1
```
(REMOTE_CONTROLBOARD_OPTIONS (writeStrict on)) (axesNames (torso_pitch torso_roll torso_yaw l_shoulder_pitch l_shoulder_roll l_shoulder_yaw l_elbow r_shoulder_pitch r_shoulder_roll r_shoulder_yaw r_elbow l_hip_pitch l_hip_roll l_hip_yaw l_knee l_ankle_pitch l_ankle_roll r_hip_pitch r_hip_roll r_hip_yaw r_knee r_ankle_pitch r_ankle_roll)) (device remotecontrolboardremapper) (localPortPrefix "/WBT/torqueControlBalancingModel1Robot1Configuration") (remoteControlBoards ("/iCub1/torso" "/iCub1/left_arm" "/iCub1/right_arm" "/iCub1/left_leg" "/iCub1/right_leg")
```  
- iCub 2
```
(REMOTE_CONTROLBOARD_OPTIONS (writeStrict on)) (axesNames (torso_pitch torso_roll torso_yaw l_shoulder_pitch l_shoulder_roll l_shoulder_yaw l_elbow r_shoulder_pitch r_shoulder_roll r_shoulder_yaw r_elbow l_hip_pitch l_hip_roll l_hip_yaw l_knee l_ankle_pitch l_ankle_roll r_hip_pitch r_hip_roll r_hip_yaw r_knee r_ankle_pitch r_ankle_roll)) (device remotecontrolboardremapper) (localPortPrefix "/WBT/torqueControlBalancingModel2Robot2Configuration") (remoteControlBoards ("/iCub2/torso" "/iCub2/left_arm" "/iCub2/right_arm" "/iCub2/left_leg" "/iCub2/right_leg"))
```

diegoferigo, 2020-02-06:
The configuration seems correct, I confirm. As discussed in the previous days f2f, did you have the chance to extend the test done in https://github.com/robotology/wb-toolbox/issues/186#issuecomment-582464615 and try an Ubuntu machine with the new 2019b Matlab version?

In this way we know that the problem is not related to the Matlab version you have on your computer nor Ubuntu. The only difference therefore is MacOS, and we can limit the debugging to the portions of code (or loaded libraries) of the OS.

---

Another try, as discussed, could be trying to switch on / off ACE in YARP (I think it's on by default). On Ubuntu I know it can be done and everything should work even without ACE, I'm not sure if it can be done also on MacOS.

lrapetti, 2020-02-06:
> The configuration seems correct, I confirm. As discussed in the previous days f2f, did you have the chance to extend the test done in [#186 (comment)](https://github.com/robotology/wb-toolbox/issues/186#issuecomment-582464615) and try an Ubuntu machine with the new 2019b Matlab version?

I will procede with this tests

> Another try, as discussed, could be trying to switch on / off ACE in YARP (I think it's on by default). On Ubuntu I know it can be done and everything should work even without ACE, I'm not sure if it can be done also on MacOS.

I don't see any `YARP_USE_ACE` option in cmake

In addition, I did some check with the debugger, and I have found out that the second initialisation of `iCub2` (the succesfully one) that I was seeing in https://github.com/robotology/wb-toolbox/issues/186#issuecomment-582364289 was triggered by the call of `getRobotInterface` in the block termination
 https://github.com/robotology/wb-toolbox/blob/f110c4b641a09a4bb392161db2ab79d0afc0adbe/toolbox/library/src/SetReferences.cpp#L276-L282

traversaro, 2020-02-06:
> > Another try, as discussed, could be trying to switch on / off ACE in YARP (I think it's on by default). On Ubuntu I know it can be done and everything should work even without ACE, I'm not sure if it can be done also on MacOS.
> 
> I don't see any `YARP_USE_ACE` option in cmake

The option is `SKIP_ACE` .

diegoferigo, 2020-02-06:
> In addition, I did some check with the debugger, and I have found out that the second initialisation of iCub2 (the succesfully one) that I was seeing in #186 (comment) was triggered by the call of getRobotInterface in the block termination

Good catch! Thanks GDB :)

lrapetti, 2020-02-06:
@diegoferigo 
> Good catch! Thanks ~~GDB~~ LLDB :)

lrapetti, 2020-02-06:
> The option is `SKIP_ACE` .

The compilation is failing when `SKIP_ACE` is on.

```
[  2%] Building CXX object src/libYARP_os/src/CMakeFiles/YARP_os.dir/yarp/os/Carrier.cpp.o
In file included from /Users/lorenzorapetti/Software/robotology-superbuild/robotology/YARP/src/libYARP_os/src/yarp/os/Carrier.cpp:11:
In file included from /Users/lorenzorapetti/Software/robotology-superbuild/robotology/YARP/src/libYARP_os/src/yarp/os/impl/TcpFace.h:17:
/Users/lorenzorapetti/Software/robotology-superbuild/robotology/YARP/src/libYARP_os/src/yarp/os/impl/TcpAcceptor.h:22:1: error: 
      Cannot implement TcpAcceptor on this platform
YARP_COMPILER_ERROR(Cannot implement TcpAcceptor on this platform)
^
/Users/lorenzorapetti/Software/robotology-superbuild/build/robotology/YARP/src/libYARP_conf/src/yarp/conf/system.h:109:35: note: 
      expanded from macro 'YARP_COMPILER_ERROR'
  #define YARP_COMPILER_ERROR(x)  __YARP_DO_PRAGMA(GCC error #x)
                                  ^
/Users/lorenzorapetti/Software/robotology-superbuild/build/robotology/YARP/src/libYARP_conf/src/yarp/conf/system.h:106:31: note: 
      expanded from macro '__YARP_DO_PRAGMA'
  #define __YARP_DO_PRAGMA(x) _Pragma (#x)
                              ^
<scratch space>:210:6: note: expanded from here
 GCC error "Cannot implement TcpAcceptor on this platform"
     ^
In file included from /Users/lorenzorapetti/Software/robotology-superbuild/robotology/YARP/src/libYARP_os/src/yarp/os/Carrier.cpp:11:
/Users/lorenzorapetti/Software/robotology-superbuild/robotology/YARP/src/libYARP_os/src/yarp/os/impl/TcpFace.h:51:21: error: 
      no type named 'TcpAcceptor' in namespace 'yarp::os::impl'
    yarp::os::impl::TcpAcceptor peerAcceptor;
    ~~~~~~~~~~~~~~~~^
/Users/lorenzorapetti/Software/robotology-superbuild/robotology/YARP/src/libYARP_os/src/yarp/os/Carrier.cpp:130:12: error: 
      cannot initialize return object of type 'yarp::os::Face *' with an rvalue
      of type 'yarp::os::impl::TcpFace *'
    return new yarp::os::impl::TcpFace();
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
3 errors generated.
make[2]: *** [src/libYARP_os/src/CMakeFiles/YARP_os.dir/yarp/os/Carrier.cpp.o] Error 1
make[1]: *** [src/libYARP_os/src/CMakeFiles/YARP_os.dir/all] Error 2
make: *** [all] Error 2
```

diegoferigo, 2020-02-06:
Ok, in MacOS you must have ACE. It was worth trying.

lrapetti, 2021-03-03:
> Further investigation should be done in order to understand if the problem is related to Matlab version or OS.

Jumping back in this old issue, I have now worked for some time on Simulink models with multiple remotecontrolboard with different OS and Matlab version. I keep experiencing the same problem on MacOS, while it never happened on Ubunt.
Indeed, I think we can say the problem is OS-related, and I have update the title accordingly.

lrapetti, 2021-03-03:
Actually, I think the same problem is happening also in presence of a single robot but with "heavy" models.

Today I was running a model which attaches to the control-board of a single icub running in simulation, runs a state-machine and a QP-based controller, and opens other yarp port using WB-toolbox block, and I was getting the following error in the Simulink diagnostic, which seems same as what observed [above](https://github.com/robotology/wb-toolbox/issues/186#issue-557603298).
```
...
INFO] |yarp.device.remote_controlboard| RemoteControlBoard is ENABLING the writeStrict option for all commands
    [INFO] |yarp.os.Port| Port /WBT/MomentumBasedControllerModel1Configuration/icubSim/right_leg/rpc:o active at tcp://10.255.111.45:10070/
    [INFO] |yarp.os.impl.NameClient| No connection to nameserver
    [INFO] |yarp.os.impl.NameClient| *** try running: yarp detect ***
    [INFO] |yarp.os.Port| Port /WBT/MomentumBasedControllerModel1Configuration/icubSim/right_leg/command:o active at tcp://10.255.111.45:10071/
    [INFO] |yarp.os.impl.NameClient| No connection to nameserver
    [INFO] |yarp.os.impl.NameClient| *** try running: yarp detect ***
    [INFO] |yarp.os.Port| Port /WBT/MomentumBasedControllerModel1Configuration/icubSim/right_leg/stateExt:i active at tcp://10.255.111.45:10072/
    [INFO] |yarp.os.impl.NameClient| No connection to nameserver
    [INFO] |yarp.os.impl.NameClient| *** try running: yarp detect ***
    [ERROR] |yarp.device.remote_controlboard| Problem connecting to /icubSim/right_leg/rpc:i, is the remote device available?
    [INFO] |yarp.os.impl.NameClient| No connection to nameserver
    [INFO] |yarp.os.impl.NameClient| *** try running: yarp detect ***
    [ERROR] |yarp.device.remote_controlboard| Problem connecting to /icubSim/right_leg/command:i, is the remote device available?
    [INFO] |yarp.os.impl.NameClient| No connection to nameserver
    [INFO] |yarp.os.impl.NameClient| *** try running: yarp detect ***
    [ERROR] |yarp.device.remote_controlboard| Problem connecting to /icubSim/right_leg/stateExt:o, is the remote device available?
    [ERROR] |yarp.dev.PolyDriver| Driver <remote_controlboard> was found but could not open
    [ERROR] |yarp.device.remotecontrolboardremapper| Opening remote_controlboard with remote " /icubSim/right_leg ", opening the device failed.
    [ERROR] |yarp.dev.PolyDriver| Driver <remotecontrolboardremapper> was found but could not open
...
```

Here are the full logs:
- [simulink_diagnostic.txt](https://github.com/robotology/wb-toolbox/files/6077762/simulink_diagnostic.txt)
- [matlab_terminal.txt](https://github.com/robotology/wb-toolbox/files/6077763/matlab_terminal.txt)
- [yarp_name_list.txt](https://github.com/robotology/wb-toolbox/files/6077766/yarp_name_list.txt)
- [yarp_server.txt](https://github.com/robotology/wb-toolbox/files/6077767/yarp_server.txt)

@nunoguedelha I don't know if you followed the discussion back in time. As a MacOS user, have you ever experienced this problem?

nunoguedelha, 2021-05-08:
Hi @lrapetti , just saw your comment now, sorry. I had indeed an issue last year, somewhat similar, not sure it's the same though. I'll send you the link. We can discuss on Monday.

lrapetti, 2021-05-10:
> Hi @lrapetti , just saw your comment now, sorry. I had indeed an issue last year, somewhat similar, not sure it's the same though. I'll send you the link. We can discuss on Monday.

Thanks @nunoguedelha, let me know if you want to have a chat regarding this issue