robotology/wb-toolbox #189: Optimize BlockInitialization MATLAB function to reduce model compilation time

traversaro, 2020-06-13:
#### Summary
It seems that the BlockInitialization MATLAB function takes a lot of  time to execute in some models, impacting a lot the time necessary to start a simulation. 

#### Motivation

Profiling in some Simulink models that use WB-Toolbox revealed that during Simulink model compilation ( https://de.mathworks.com/help/simulink/ug/simulating-dynamic-systems.html#f7-8245 ) .  In particular, it seems that the part that costs more in term of time is the following  loop: https://github.com/robotology/wb-toolbox/blob/114ac0e76068399c51150cb55c5922634d199ad3/matlab/%2BWBToolbox/BlockInitialization.m#L30-L44, that takes  ~100 ms for each iteration, and perform an interation for each nesting level of the block, arriving to a time ~1/2 seconds for really nested models, and a total compilation time of ~1 minute for a model with a lot of WB-Toolbox blocks. 

#### Additional context

Further profiling to understand  which line of https://github.com/robotology/wb-toolbox/blob/114ac0e76068399c51150cb55c5922634d199ad3/matlab/%2BWBToolbox/BlockInitialization.m#L30-L44 takes so much time may be needed.

traversaro, 2020-06-13:
> Further profiling to understand  which line of https://github.com/robotology/wb-toolbox/blob/114ac0e76068399c51150cb55c5922634d199ad3/matlab/%2BWBToolbox/BlockInitialization.m#L30-L44 takes so much time may be needed. 

I did not profile those blocks, so it is possible that I may be wrong, but **under the hypothesis that the bottleneck is the call to `regexp`**, perhaps  a simpler  strategy to identify the block strings that start with the value of `analyzedScope` and end with `/ImConfig` may  be used. For example, just checking if the string startsWith the value  `analyzedScope`  (via https://de.mathworks.com/help/matlab/ref/startswith.html) and ends with the `/ImConfig` (via https://de.mathworks.com/help/matlab/ref/endswith.html) may be more efficient.

traversaro, 2020-06-13:
cc @Giulero @diegoferigo @nunoguedelha @lrapetti @GiulioRomualdi  @S-Dafarra

diegoferigo, 2020-06-13:
Thanks for opening the issue and providing a recap of all what we discussed, this is very helpful!

> I did not profile those blocks, so it is possible that I may be wrong, but under the hypothesis that the bottleneck is the call to regexp, perhaps a simpler strategy to identify the block strings that start with the value of analyzedScope and end with /ImConfig may be used (if that is the goal of that line of code, it is not clear to me at the first glance, perhaps @diegoferigo can provide some comments on this).

I wrote this piece of code quite a lot of time ago and I don't really remember details :) Back in time, our Simulink models were not as complex as those we're using right now and optimizing this first implementation was not a priority.

A given depth of the model's tree (let's assume the second level) has a config block if the following is found:

```
/firstlevel/secondlevel/<config_block_name>/ImConfig
```

The block name is up to the user, we cannot assume is always the default `Config`. The `Config` block is just a masked subsystem containing all the information we need in the mask. No more no less. In order to implement the lookup logic, I added a dummy `ImConfig` block in this subsystem (you can look under the mask of the block if you're curious).

The regexp looks for exactly this pattern. If I recall, the caveat here is that between `secondlevel/` and `/ImConfig` there should be no `/`, otherwise the level is changed and blocks in deeper levels would match (think of `/firstlevel/secondlevel/thirdlevel/<config_block_name>/ImConfig`). The simplified method you described that uses `{starts,ends}with` should further check that the remaining part of the string contains no `/`.

traversaro, 2020-06-13:
Just a small doubt: what does the `Im` in `ImConfig` mean?

diegoferigo, 2020-06-13:
I'm config xD

traversaro, 2020-06-13:
> I'm config xD

Ack. If a long scoped block name is not a problem, for the next release of WB-Toolbox in which we break the compatibility it probably could be more safe to use a more project specific name (something like `WBToolboxImConfig`) to avoid conflicts with other projects.

diegoferigo, 2020-06-13:
Sure!

nunoguedelha, 2020-06-14:
So, as @diego mentioned, wouldn't the 3 actions:
- check if the string starts with the value `analyzedScope`,
- check if the string ends with the value `ImConfig`,
- check if there is no `/` in between,

be as expensive as the Regexp, which is doing exactly the same thing?

Anyway, maybe just reducing the list of the blocks over which we are searching for the pattern `analyzedScope/<config_block_name>/ImConfig` would significantly improve the speed. Typically instead of the line below:
https://github.com/robotology/wb-toolbox/blob/114ac0e76068399c51150cb55c5922634d199ad3/matlab/%2BWBToolbox/BlockInitialization.m#L19-L20

we should just get the blocks having a `ImConfig`, and this using the regexp built-in option of `find_system`:
```
blocks_system = find_system(topLevel,'LookUnderMasks','on','FollowLinks','on','Regexp','on','Name','ImConfig');
```
In our project, this reduces the list from 2032 to 4 elements.

nunoguedelha, 2020-06-14:
Actually, with that small change, I went from 120s compilation to 10s only!

traversaro, 2020-06-14:
> be as expensive as the Regexp, which is doing exactly the same thing?

Given how Regexp work, I would expect the hardcoded version to be much faster (see https://de.mathworks.com/matlabcentral/answers/162930-why-regexprep-much-slower-than-strrep https://de.mathworks.com/matlabcentral/answers/95953-why-can-nested-quantifiers-in-regexp-can-cause-inefficient-failures-in-matlab-6-5-r13 https://de.mathworks.com/matlabcentral/answers/44700-recommended-steps-to-deal-with-a-slow-loop), but in general it depends a lot on the regexp implementation and in the regexp itself, and typically is faster to check the speed directly rather then speculate. 

>  we should just get the blocks having a ImConfig, and this using the regexp built-in option of find_system:

If `ImConfig` is always the last part of the System name string, perhaps it can be helpful for the regexp engine to change the regexp to `ImConfig$`, so that it matches only system names that end with `ImConfig` ? A similar optimization probably can also be done with the later regexp.

> Actually, with that small change, I went from ~50s compilation to ~10s only!

Great! What is the bottleneck  now? Because probably it is something that can be further improved.

diegoferigo, 2020-06-14:
Awesome @nunoguedelha! I suspect that using the termination constraint `$` as @traversaro wrote could be even faster.

Did you by chance tried to check if it keeps working also in a multi-config model? Each block that requires the config block also has in its mask a button that highlights the config block it found. In this way you can easily check that a multi-robot setup would not get broken.

nunoguedelha, 2020-06-14:
Indeed @diegoferigo , that's a good point to check. I expect it to work since I didn't change the loop that looks for a Config block going up in the hierarchy. I just removed from the list all the lines that in any case would never be a possible match.

Anyway I checked, it works fine. Two different WBT blocks find properly their respective Config blocks. In the case where there is a Config at the root level, and several nested Config blocks, the closest one is always the one found when going up in the hierarchy.

nunoguedelha, 2020-06-14:
CC @CarlottaSartore

DanielePucci, 2020-06-14:
Probably, also @bemilio would be interested in this activity

Giulero, 2020-06-15:
Your hypothesis about `regexp` seems to be confirmed.
https://github.com/dic-iit/element_ironcub-control/issues/127#issuecomment-644047678

nunoguedelha, 2020-06-15:
## Optimizing the regexp pattern

> If `ImConfig` is always the last part of the System name string, perhaps it can be helpful for the regexp engine to change the regexp to `ImConfig$`,

Good point, unfortunately I tried the optimization using `ImConfig$` instead of `ImConfig` and didn't see any significant improvement. I've used tic & toc to measure min,max,mean,std of the time spent on each code sector:
- block 1 : 1rst regexp (lines 10 to 28)
- block 2 : 2nd regexp (lines 31 to 44)
- block 3 : read the parameters from the mask (lines 46 to 63)

Here are the results in seconds...

With `ImConfig`:
```
block 1 delay MIN : 0.00088578
block 1 delay MAX : 0.012578
block 1 delay MEAN: 0.0068732 <----
block 1 delay STD : 0.0019178
block 2 delay MIN : 0.00040019
block 2 delay MAX : 0.0042012
block 2 delay MEAN: 0.0017049 <----
block 2 delay STD : 0.00080383
block 3 delay MIN : 0.0025003
block 3 delay MAX : 0.016778
block 3 delay MEAN: 0.0035257 
block 3 delay STD : 0.0019485
total init delay is : 0.82306
number of WBT blocks : 68
init delay MIN : 0.0088708
init delay MAX : 0.020219
init delay MEAN: 0.012104
init delay STD : 0.0026661
```

With `ImConfig$` on both regexp:
```
block 1 delay MIN : 0.00093944
block 1 delay MAX : 0.012867
block 1 delay MEAN: 0.008411 <----
block 1 delay STD : 0.0024316
block 2 delay MIN : 0.00041072
block 2 delay MAX : 0.0051588
block 2 delay MEAN: 0.0016555 <----
block 2 delay STD : 0.00081421
block 3 delay MIN : 0.002502
block 3 delay MAX : 0.014411
block 3 delay MEAN: 0.0034467
block 3 delay STD : 0.0017981
total init delay is : 0.9189
number of WBT blocks : 68
init delay MIN : 0.009108
init delay MAX : 0.021265
init delay MEAN: 0.013513
init delay STD : 0.0029967
```
With exact word match `\<ImConfig\>` on both regexp:
```
block 1 delay MIN : 0.00087865
block 1 delay MAX : 0.014009
block 1 delay MEAN: 0.0085083 <----
block 1 delay STD : 0.0025692
block 2 delay MIN : 0.00042499
block 2 delay MAX : 0.0034771
block 2 delay MEAN: 0.0015776 <----
block 2 delay STD : 0.00060321
block 3 delay MIN : 0.0025053
block 3 delay MAX : 0.016583
block 3 delay MEAN: 0.0034615
block 3 delay STD : 0.0019693
total init delay is : 0.92122
number of WBT blocks : 68
init delay MIN : 0.0093781
init delay MAX : 0.022289
init delay MEAN: 0.013547
init delay STD : 0.0031138
```
so to recap the mean delays ...
| | `ImConfig` |  `ImConfig$` | `\<ImConfig\>` |
| --- | --- | --- | --- |
| block 1 | 0.0068732 | 0.008411 | 0.0085083 |
| block 2 | 0.0017049 | 0.0016555 | 0.0015776 |

So it's even slower with `ImConfig$` and `\<ImConfig\>`.

> A similar optimization probably can also be done with the later regexp.

Actually If we use the regexp in the first code block (when doing the `find_system`), all the elements in the list have `ImConfig`, so it shouldn't matter.

## First compilation after Matlab launch VS subsequent compilations

First compilation after starting Matlab:
```
number of WBT blocks : 68
total init delay is : 0.79796
```

Subsequent compilations:
```
number of WBT blocks : 68
total init delay is : 0.85303
```

The results are in seconds. There's no significant change between the two cases.

nunoguedelha, 2020-06-15:
> Your hypothesis about `regexp` seems to be confirmed.
> [image]

Nice stats report @Giulero ! could you run that again with the change mentioned in https://github.com/robotology/wb-toolbox/issues/189#issuecomment-643708220 ?
```
blocks_system = find_system(topLevel,'LookUnderMasks','on','FollowLinks','on','Regexp','on','Name','ImConfig');
```

Giulero, 2020-06-15:
@nunoguedelha with this change the compilation time is reduced to 4 seconds!
These stats are relative to the `BlockInitialization`
[image]

nunoguedelha, 2020-06-15:
btw @Giulero , how did you get these stats line by line?

traversaro, 2020-06-15:
> @nunoguedelha with this change the compilation time is reduced to 4 seconds!
> These stats are relative to the `BlockInitialization`
> [image]

Perfect, then if the `Blockinitialization` just takes 0.5 seconds out of 4 seconds, problably it is  already worth to open a PR with @nunoguedelha modification.

nunoguedelha, 2020-06-15:
It's probably not worth it, but actually I also tried another optimization (using a persistent variable for storing the system blocks list) that devides by 10 the execution time of the `find_system` section (first regexp). In my machine, the `Blockinitialization` went from 0.9s to 0.4s total execution time.

I can submit the PR with the initial change.

Giulero, 2020-06-15:
> btw @Giulero , how did you get these stats line by line?

@nunoguedelha it is super easy! Have a look here https://it.mathworks.com/help/matlab/matlab_prog/profiling-for-improving-performance.html

lrapetti, 2020-06-16:
Also in my case performance improved drastically. I tested in on two different controllers (in both there are two robots) on my laptop (MacOS Mojave).

### Before
- controller 1
[image]
[image]
- controller 2
[image]
[image]

### After
- controller 1
[image]
[image]
- controller 2
[image]
[image]

DanielePucci, 2020-06-19:
Guys, can we close this issue then?

CC
@diegoferigo @traversaro

traversaro, 2020-06-19:
Sure!

closed 2020-06-19 by traversaro