robotology/wb-toolbox #192: Failure when two QP blocks - one with inequalities and the other not - are used in the same model

gabrielenava, 2020-06-30:
- [x] I already updated to the latest version I can use
- [x] I already checked similar issues using the search box
- [x] I already checked the website for known troubleshooting
- [x] I already cleaned my environment (by removing the wb-toolbox install folder and installing it again)
- [x] I already excluded the possibility that system configuration files (e.g. yarp related) might be responsible
- [x] I am sure the environment variables are correct
- [x] I tried to reproduce the error in simulation (if it affects the real robot)

#### Description

When two QP blocks are added to the same Simulink model, and one of the two uses both input boundaries and inequality constraints while the other uses only input boundaries, the model exits with an error right after startup.

#### Platform

iRonCub controller, simple Simulink example (attached to this issue).

#### Reproducibility

Here is a simple Simulink model I created to reproduce the bug: 

[testQPblock.zip](https://github.com/robotology/wb-toolbox/files/4853592/testQPblock.zip)

#### Screenshots

Here is an overview of the simple Simulink model

[image]
[image]
[image]

Here is the error

[image]

the error does not appear if I run only 1 QP and I comment the other, or if I run QPTest 1 (unconstrained) and QPTest 3 or QPTest 1 and QPTest 2.

 
#### Additional information

- OS: Ubuntu 18.04 LTS
- Version / Commit: https://github.com/robotology/wb-toolbox/commit/0688771c6124a56fb975f89f50b697f3410282ad
- Matlab Version: 2019b

diegoferigo, 2020-07-01:
Thanks @gabrielenava for filling this issue. The qpOASES block has quite a convoluted logic for its port due to its high number of signal. Things are even more complicated since all its signals are dynamically sized.

On the fly, I cannot explain why the blocks do not work if they are both enabled in the model. I suspect that a **1D signal is treated as a vector instead of a matrix with just one row** (or the other way around). Can you please provide in all the combinations few screenshots with the signal dimensions displayed (you can find the right button in the menu)?

As a reference, here below you can find the logic that defines input port dimensions:

https://github.com/robotology/wb-toolbox/blob/0688771c6124a56fb975f89f50b697f3410282ad/toolbox/library/src/QpOases.cpp#L145-L203

gabrielenava, 2020-07-01:
Sure, here are the screenshots with the signal dimensions:

**QP with inputs bounds only**

[image]

**QP with inputs bounds and inequalities**

[image]

I also drop again the error message, because I think the message got cut in the above comment while now I see the full error:

[image]

diegoferigo, 2020-07-01:
Can you replace the all blocks from e.g. `2*ones(2,1)` to `2 * [1; 1]`? Try to get a matrix as output.

gabrielenava, 2020-07-01:
I did it for every element:

[image]

But it didn't solve the issue. The error is the same.

diegoferigo, 2020-07-02:
Mmh ok it was worth trying. Breaking down the error message, it seems that the stack trace is the following:

1. `Unsupported number of port dimensions for port at index` from [SimulinkBlockInformationImpl.cpp#L289-L291](https://github.com/robotology/blockfactory/blob/4cbac726a49f6ca9a168c1d3578fd9df5a4861ed/sources/Simulink/src/SimulinkBlockInformationImpl.cpp#L289-L291)
1. `Input port at index 5does not contain a vector. Failed to get its size.` from [SimulinkBlockInformation.cpp#L49-L62](https://github.com/robotology/blockfactory/blob/4cbac726a49f6ca9a168c1d3578fd9df5a4861ed/sources/Simulink/src/SimulinkBlockInformation.cpp#L49-L62)
1. `Sizes of bounds do not match with the number of variables.` from [QpOases.cpp#L266-L277](https://github.com/robotology/wb-toolbox/blob/0688771c6124a56fb975f89f50b697f3410282ad/toolbox/library/src/QpOases.cpp#L266-L277)

The origin of the error seems to be 1, somehow the call of `ssGetInputPortNumDimensions(simstruct, idx)` in the switch statement does not return `1` nor `2` (for, respectively, vector and matrix signals). Can you try to print its value in the two cases with only one QpOases block and with two of them?

gabrielenava, 2020-07-02:
Ok I'll try. I also noted one thing I beleive is important: the error message is generated when processing QP Test 2, that in my example is the QP block **without inequality constraints**, so it has a total of 4 input ports. However the error is generated when trying to process **input port with index 5**, which does not exist in that QP block.

diegoferigo, 2020-07-02:
Good catch, this was not clear. Can you check under the mask if the S-Function is properly connected in the block that fails with only 4 inputs? I suspect that there's something strange going on there.

gabrielenava, 2020-07-02:
I printed a message to check the output of `ssGetInputPortNumDimensions(simstruct, idx)` which is used in the switch-case logic.

Here is the result:

[image]

so it exits almost immediately after detecting a port with input 0. If I comment out TEST QP3, the imput ports are always either of dimension 1 or 2 (even if I don't know why the check is called so many times/ the simulation lasts 1 single step time):

[image]

this is the result of pressing: `look under the mask` on the QP block:

[image]

nunoguedelha, 2021-05-03:
@diegoferigo @gabrielenava , I was happy to find this issue. I ran into another use case which is also failing, in this [model](https://github.com/robotology/whole-body-controllers/tree/feature/replace-gazebo-by-matlab-simulator/controllers/floating-base-balancing-torque-control-with-simulator).

The model is based on [floating-base-balancing-torque-control](https://github.com/robotology/whole-body-controllers/tree/devel/controllers/floating-base-balancing-torque-control), but the configuration of the QP blocks were the same: **[H, g, A, ubA] inputs** for both blocks. The issue disappeared after adding the input "lbA" to both blocks.

Considering the similar issue https://github.com/dic-iit/element_ironcub-control/issues/164#issuecomment-821027420, it looks like the QP block is likely to fail if we use any but the following configurations:
- unconstrained:  H, g
- bounded constraints: H, g, A, lbA, ubA
- bounded constraints and variable: H, g, A, lbA, ubA, lb, ub

**So the QP block logic checking the ports dimensions doesn't handle well leaving holes in the port list, i.e. "skipping" ports, if a QP block with all the inputs H, g, A, lbA, ubA, lb, ub also exists.**

As per this hypothesis, the following will always work:
- H, g, A, lbA
- H, g, A, lbA, ubA, lb

... and the following will fail, unless no fully activated QP block exists in the same model:
- H, g, A, (missing lbA), ubA
- H, g, (missing A, lbA, ubA), lb, ub

nunoguedelha, 2021-05-03:
@diegoferigo , @traversaro, I hope you don't mind, I added this to the board of element software engineering (next sprint tag), just for tracking it closer.

diegoferigo, 2021-05-03:
> @diegoferigo , @traversaro, I hope you don't mind, I added this to the board of element software engineering (next sprint tag), just for tracking it closer.

As long as you remember that this repo is public and all links to private repos cannot be reached by external users (like the element you linked above), is fine for me :)

> So the QP block logic checking the ports dimensions doesn't handle well leaving holes in the port list, i.e. "skipping" ports, if a QP block with all the inputs H, g, A, lbA, ubA, lb, ub also exists.

I remember having quite a lot of problems with the signal dimensions, that required dirty workarounds like adding in the simulink side extra blocks that were forcing some of the output signals sizes match some of the inputs sizes. I don't have on the fly a working Matlab setup, but if you look under the mask of the block you will surely understand what I mean (if you can, please also attach a screenshot in this issue).

That workaround was working fine in the Matlab version I had at that time, I won't be surprised if it does not work reliably on new versions.