robotology/wb-toolbox #197: Error using WBToolbox on Windows on MATLAB R2019b

traversaro, 2021-01-21:
#### Description

The WBToolbox blocks do not load correctly on Windows. 

#### Platform

Windows on Simulink, no robot involved. 

#### Reproducibility

Unfortunately, this is emerged in a private repo, but I guess you can reproduce by running the https://github.com/robotology/whole-body-controllers on Windows.

#### Additional information

- OS: Windows
- Visual Studio: 2019 v16.8
- Version / Commit: v5.1/ BlockFactory v0.8.1 
- Matlab Version: 2019b

#### Additional context

The error was experienced by @vpunithreddy . The first error was:
~~~
Error evaluating 'InitFcn' callback of SubSystem block 'PositionControl/Simulator1'. 
Callback string is '%% initialize the simulation
cd(fileparts(which(bdroot)));
initSimulatorCallback;'
Caused by:
Error due to multiple causes. 
Error reported by S-function 'BlockFactory' in 'initSimulator/Transformation Matrix From FixedLink to Base/RelativeTransform Between l_sole and Base/S-Function':
Failed to create factory
Factory error (2): The specified module could not be found.

Failed to get factory object (className=ForwardKinematics,libName=WBToolbox)
Failed to get the factory 
Error reported by S-function 'BlockFactory' in 'initSimulator/l_sole/S-Function':
Failed to create factory
Factory error (2): The specified module could not be found.

Failed to get factory object (className=ForwardKinematics,libName=WBToolbox)
Failed to get the factory 
Error reported by S-function 'BlockFactory' in 'initSimulator/r_sole/S-Function':
Failed to create factory
Factory error (2): The specified module could not be found.

Failed to get factory object (className=ForwardKinematics,libName=WBToolbox)
Failed to get the factory 
Component:Simulink | Category:Model error
~~~

Even if the name/error is not clear, the error code (2) pointed out that for some reason the `WBToolbox.dll` plugin was not found (see https://github.com/robotology/blockfactory/blob/v0.8.1/deps/sharedlibpp/src/shlibpp/SharedLibraryFactory.h#L41). However, this was an error in setting `BLOCKFACTORY_PLUGIN_PATH` , that was fixed by https://github.com/robotology/robotology-superbuild/pull/599 . After fixing that, we run again the model and we obtained:
~~~
Error evaluating 'InitFcn' callback of SubSystem block 'PositionControl/Simulator1'. 
Callback string is '%% initialize the simulation
cd(fileparts(which(bdroot)));
initSimulatorCallback;'
Caused by:
Error due to multiple causes. 
Error reported by S-function 'BlockFactory' in 'initSimulator/Transformation Matrix From FixedLink to Base/RelativeTransform Between l_sole and Base/S-Function':
Failed to create factory
Factory error (3): The specified procedure could not be found.

Failed to get factory object (className=ForwardKinematics,libName=WBToolbox)
Failed to get the factory 
Error reported by S-function 'BlockFactory' in 'initSimulator/l_sole/S-Function':
Failed to create factory
Factory error (3): The specified procedure could not be found.

Failed to get factory object (className=ForwardKinematics,libName=WBToolbox)
Failed to get the factory 
Error reported by S-function 'BlockFactory' in 'initSimulator/r_sole/S-Function':
Failed to create factory
Factory error (3): The specified procedure could not be found.

Failed to get factory object (className=ForwardKinematics,libName=WBToolbox)
Failed to get the factory 
~~~

So now the error is instead that the library was not loaded (see https://github.com/robotology/blockfactory/blob/v0.8.1/deps/sharedlibpp/src/shlibpp/SharedLibraryFactory.h#L42).

traversaro, 2021-01-25:
I tested on Windows with robotology-superbuild-vcpkg-dependencies 0.6.0 and I could not replicate this behaviour, as the blocks are loaded correctly. However, I added in Blockfactory a simple command line tool to simplify test and debug of situation like that, in https://github.com/robotology/blockfactory/pull/61 . 

In particular, if you like to debug this situation further @vpunithreddy I would kindly ask to do the following:
~~~
# Switch to the branch of the blockfactory repo 
cd ~/Documents/GitHub/robotology-superbuild/robotology/BlockFactory
git fetch
git checkout add-exists-tool
# Re-compiled blockfactory
cd ../../build/robotology/BlockFactory
cmake --build . --config Release --target INSTALL
~~~

At this point, you should have a new command line tool called `blockfactory-exists`, that should be able to check if a given block exists. In particular, I would be curious of the output of:
~~~
blockfactory-exists WBToolbox ForwardKinematics
~~~
On my system in which everything is working fine, the output is:
~~~
SUCCESS: Block "ForwardKinematics" found and loaded from plugin "WBToolbox".
~~~
on your system given your output I would guess that the output should be different.

vpunithreddy, 2021-02-11:
I followed the procedure described by @traversaro and was able to receive the message
` SUCCESS: Block "ForwardKinematics" found and loaded from plugin "WBToolbox"
I shall check if the error still exists for simulink

traversaro, 2021-03-11:
@vpunithreddy Now that we have Conda binaries for WB-toolbox also on Windows the installation process should be much simpler, in a nutshell you should just follow https://github.com/robotology/robotology-superbuild/blob/2b0d3299ffecc3f4bce780ae999b08d2432d59d0/doc/conda-forge.md#binary-installation, and then in a command prompt in which you enabled your environment, run:
~~~
conda install -c robotology-staging wb-toolbox
~~~
and then from the same terminal launch matlab. You should not even have either CMake or Visual Studio installed.

traversaro, 2021-05-18:
I just installed wb-toolbox from source with conda-provided dependencies, and I have a similar problem when using the QP block:
~~~
Error reported by S-function 'BlockFactory' in 'qpoases_vs_osqp_test/QP':
Failed to create factory
Factory error (4): The specified procedure could not be found.

Failed to get factory object (className=QpOases,libNam��m-L
~~~

traversaro, 2021-05-18:
I simply had forgot to install qpoases initially, so the initial value of `WBT_USES_QPOASES` was set to false. After I installed `qpoases`:
~~~
conda install qpoases
~~~
enabled `WBT_USES_QPOASES`  and recompiled everything, everything was working as expected:
[image]

traversaro, 2022-02-21:
Closing as the problem was solved.

closed 2022-02-21 by traversaro