robotology/wb-toolbox #199: Error using WBToolbox on Ubuntu 20.04

S-Dafarra, 2021-01-25:
- [X] I already updated to the latest version I can use
- [X] I already checked similar issues using the search box (similar, but in Windows 
- [X] I already checked the website for known troubleshooting
- [X] I already cleaned my environment (by removing the wb-toolbox install folder and installing it again)
- [X] I already excluded the possibility that system configuration files (e.g. yarp related) might be responsible
- [X] I am sure the environment variables are correct
- [X] I tried to reproduce the error in simulation (if it affects the real robot)

#### Description

When I try to compile a simple projects like https://github.com/robotology/whole-body-controllers/blob/master/utilities/debug_positionControl.mdl, it fails with the errors:
```
Failed to create factory
Factory error (3): /usr/local/MATLAB/R2020a/bin/glnxa64/../../sys/os/glnxa64/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /home/sdafarra/Software/robotology-superbuild/build/install/lib/libidyntree-high-level.so)
Failed to get factory object (className=YarpRead,libName=WBToolbox)
Failed to get the factory
```
#### Platform

Ubuntu 20.04. It happens also without the robot or a simulator. Simply open the model of the link and try to compile.

#### Reproducibility

Describe the steps for reproducing the behavior. If possible, upload a simple Simulink model (only if the error does not either involve the real robot or it is reproducible in simulation).

1. Open https://github.com/robotology/whole-body-controllers/blob/master/utilities/debug_positionControl.mdl
2. Compile

#### Additional information

- OS: Ubuntu 20.04
- Version / Commit: 99156c73c6b651502317722dc908ba7a9ba5f4ce (Current master)
- Matlab Version: 2020a

#### Additional context

It was working just months ago, on the same system. I wonder what it could have gone wrong. I usually call matlab as follows 
```
alias matlab='cd ~/Documents/MATLAB && LD_PRELOAD=$LD_PRELOAD:/usr/lib/x86_64-linux-gnu/libstdc++.so.6 && LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$SOFTWARE_DIR/IPOPT_Solvers matlab'
```
I wonder if this is still [TROMH](https://github.com/robotology/wb-toolbox/issues/101#issuecomment-372016880) hitting once more.

<details>
<summary>Long output </summary>

```
Failed to create factory
Factory error (3): /usr/local/MATLAB/R2020a/bin/glnxa64/../../sys/os/glnxa64/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /home/sdafarra/Software/robotology-superbuild/build/install/lib/libidyntree-high-level.so)
Failed to get factory object (className=YarpRead,libName=WBToolbox)
Failed to get the factory
Failed to create factory
Factory error (3): /usr/local/MATLAB/R2020a/bin/glnxa64/../../sys/os/glnxa64/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /home/sdafarra/Software/robotology-superbuild/build/install/lib/libidyntree-high-level.so)
Failed to get factory object (className=YarpRead,libName=WBToolbox)
Failed to get the factory
Failed to create factory
Factory error (3): /usr/local/MATLAB/R2020a/bin/glnxa64/../../sys/os/glnxa64/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /home/sdafarra/Software/robotology-superbuild/build/install/lib/libidyntree-high-level.so)
Failed to get factory object (className=YarpRead,libName=WBToolbox)
Failed to get the factory
Failed to create factory
Factory error (3): /usr/local/MATLAB/R2020a/bin/glnxa64/../../sys/os/glnxa64/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /home/sdafarra/Software/robotology-superbuild/build/install/lib/libidyntree-high-level.so)
Failed to get factory object (className=YarpRead,libName=WBToolbox)
Failed to get the factory
Failed to create factory
Factory error (3): /usr/local/MATLAB/R2020a/bin/glnxa64/../../sys/os/glnxa64/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /home/sdafarra/Software/robotology-superbuild/build/install/lib/libidyntree-high-level.so)
Failed to get factory object (className=YarpRead,libName=WBToolbox)
Failed to get the factory
Failed to create factory
Factory error (3): /usr/local/MATLAB/R2020a/bin/glnxa64/../../sys/os/glnxa64/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /home/sdafarra/Software/robotology-superbuild/build/install/lib/libidyntree-high-level.so)
Failed to get factory object (className=YarpRead,libName=WBToolbox)
Failed to get the factory
Failed to create factory
Factory error (3): /usr/local/MATLAB/R2020a/bin/glnxa64/../../sys/os/glnxa64/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /home/sdafarra/Software/robotology-superbuild/build/install/lib/libidyntree-high-level.so)
Failed to get factory object (className=YarpRead,libName=WBToolbox)
Failed to get the factory
Failed to create factory
Factory error (3): /usr/local/MATLAB/R2020a/bin/glnxa64/../../sys/os/glnxa64/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /home/sdafarra/Software/robotology-superbuild/build/install/lib/libidyntree-high-level.so)
Failed to get factory object (className=YarpRead,libName=WBToolbox)
Failed to get the factory
Failed to create factory
Factory error (3): /usr/local/MATLAB/R2020a/bin/glnxa64/../../sys/os/glnxa64/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /home/sdafarra/Software/robotology-superbuild/build/install/lib/libidyntree-high-level.so)
Failed to get factory object (className=YarpRead,libName=WBToolbox)
Failed to get the factory
Failed to create factory
Factory error (3): /usr/local/MATLAB/R2020a/bin/glnxa64/../../sys/os/glnxa64/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /home/sdafarra/Software/robotology-superbuild/build/install/lib/libidyntree-high-level.so)
Failed to get factory object (className=YarpRead,libName=WBToolbox)
Failed to get the factory
Failed to create factory
Factory error (3): /usr/local/MATLAB/R2020a/bin/glnxa64/../../sys/os/glnxa64/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /home/sdafarra/Software/robotology-superbuild/build/install/lib/libidyntree-high-level.so)
Failed to get factory object (className=GetMeasurement,libName=WBToolbox)
Failed to get the factory
Failed to create factory
Factory error (3): /usr/local/MATLAB/R2020a/bin/glnxa64/../../sys/os/glnxa64/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /home/sdafarra/Software/robotology-superbuild/build/install/lib/libidyntree-high-level.so)
Failed to get factory object (className=GetMeasurement,libName=WBToolbox)
Failed to get the factory
Failed to create factory
Factory error (3): /usr/local/MATLAB/R2020a/bin/glnxa64/../../sys/os/glnxa64/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /home/sdafarra/Software/robotology-superbuild/build/install/lib/libidyntree-high-level.so)
Failed to get factory object (className=GetMeasurement,libName=WBToolbox)
Failed to get the factory
Failed to create factory
Factory error (3): /usr/local/MATLAB/R2020a/bin/glnxa64/../../sys/os/glnxa64/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /home/sdafarra/Software/robotology-superbuild/build/install/lib/libidyntree-high-level.so)
Failed to get factory object (className=RealTimeSynchronizer,libName=WBToolbox)
Failed to get the factory
```
</details>

traversaro, 2021-01-25:
Probably some change in iDynTree, given that the error is:
> Factory error (3): /usr/local/MATLAB/R2020a/bin/glnxa64/../../sys/os/glnxa64/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /home/sdafarra/Software/robotology-superbuild/build/install/lib/libidyntree-high-level.so)

Cross-link : https://github.com/robotology/robotology-superbuild/issues/64 .

S-Dafarra, 2021-01-25:
I am using the current iDynTree master https://github.com/robotology/idyntree/commit/e52d72a6fb3c0872bd9bcf830325776caf909db5

S-Dafarra, 2021-01-25:
Related comment: https://github.com/robotology/robotology-superbuild/issues/64#issuecomment-767042608

S-Dafarra, 2021-01-25:
This got fixed by installing the [``matlab-support`` package](https://packages.ubuntu.com/focal/matlab-support). See https://github.com/robotology/robotology-superbuild/issues/64#issuecomment-727241894

closed 2021-01-25 by S-Dafarra