robotology/wb-toolbox #203: Simulation fails if run with sim command from workspace other than Base workspace

nunoguedelha, 2021-03-12:
- [x] I already updated to the latest version I can use
- [x] I already checked similar issues using the search box

#### Description

If we run a simulation programmatically from a function workspace, using the command [`sim`](https://www.mathworks.com/help/simulink/ug/using-the-sim-command.html) ...
```
myFunction()
...
function myFunction()

a = sim('initSimulator','SimulationMode','normal','SrcWorkspace','current');

end
```
... the simulation fails with the error message:
```
Error evaluating 'InitFcn' callback of SubSystem block '<some-system>'. 
Callback string is '%% initialize the simulation
cd(fileparts(which(bdroot)));
<some-init-script-from-where-myFunction-is-called>;'
Caused by:
Error evaluating 'InitFcn' callback of SubSystem block (mask) '<parent sub system path>/Configuration'. 
Callback string is 'source = get_param(gcb,'ConfigSource');

if strcmp(source, 'Workspace')
    object = get_param(gcb,'ConfigObject');
    WBToolbox.ConfigurationToMask(gcb, object);
end

WBToolbox.MaskToConfiguration(gcb);

clear object source
' 
[ConfigurationToMask] Failed to get WBTConfigRobot object from the workspace 
```

#### Platform

This can happen in simulation as when using the real robot.

#### Reproducibility

Steps for reproducing the behavior. Run the [issueExample.m]() script in matlab-whole-body-simulators, branch `debug/issue-example`.

#### Error Output

```
Error using issueExample>programmaticRun (line 6)
Error evaluating 'InitFcn' callback of SubSystem block (mask) 'test_matlab_system_2020b/Configuration'.
Callback string is 'source = get_param(gcb,'ConfigSource');

if strcmp(source, 'Workspace')
    object = get_param(gcb,'ConfigObject');
    WBToolbox.ConfigurationToMask(gcb, object);
end

WBToolbox.MaskToConfiguration(gcb);

clear object source
'

Error in issueExample (line 2)
programmaticRun(); - Show complete stack trace
Caused by:
    Error using issueExample>programmaticRun (line 6)
    [ConfigurationToMask] Failed to get WBTConfigRobotSim object from the workspace - Show complete stack trace
```

#### Additional information

- OS: macOS Catalina 10.15.7
- Version / Commit: 0db6ae4c8a5f035d70ad2e1f5c0c8762ff51ad8a (same content as v5.3)
- Matlab Version: 2020b

nunoguedelha, 2021-03-12:
## Analysis

The issue occurs in the `InitFcn` callback of the `Config` block, which calls `ConfigurationToMask`, and more specifically in line:
https://github.com/robotology/wb-toolbox/blob/4b230473809da2e56331208b862ea628b95d2660/matlab/%2BWBToolbox/ConfigurationToMask.m#L34

Here the function looks for the `WBTConfigRobotSim` (WBTConfigObjectName) systematically in the base workspace, even when we run the simulation from a function workspace and initialize all the parameters of the model in that same workspace.

## Suggested Solution

A more robust approach would be to do instead the following:
1. look for `WBTConfigObjectName` in the model workspace
2. if it is not found there, look for it in the base workspace

This way, the user designing a model can chose to define the _WBTConfigRobot_ object in the model workspace. This would be cleaner than using a workaround which copies the _WBTConfigRobot_ to the base workspace. In general, when dealing with multiple models or when running a simulation using the `sim` command, it is safer to store all variables of a specific model in the respective model workspace.

What do you think @diegoferigo  @traversaro ?

diegoferigo, 2021-03-12:
Nothing to object it the workaround you proposed maintains backward compatibility and solves the specific problem you described.

traversaro, 2021-03-12:
I think the proposed solution make sense, but perhaps people with Simulink experience may provide some input @gabrielenava @Giulero .