robotology/wb-toolbox #207: Website visualization is broken

traversaro, 2021-05-18:
Website is broken: https://robotology.github.io/wb-toolbox/mkdocs/index.html .

[image]

traversaro, 2021-05-18:
cc @diegoferigo

traversaro, 2021-07-08:
Now the job for generating the website is broken:
~~~
2021-07-08T06:47:33.1767736Z Patching output file 237/240
2021-07-08T06:47:33.1769143Z Patching output file 238/240
2021-07-08T06:47:33.1770006Z Patching output file 239/240
2021-07-08T06:47:33.1770432Z Patching output file 240/240
2021-07-08T06:47:33.1770972Z lookup cache used 983/65536 hits=7080 misses=1020
2021-07-08T06:47:33.1771416Z finished...
2021-07-08T06:47:33.1812383Z [100%] Built target dox
2021-07-08T06:47:33.2006450Z [100%] Generating doxygen documentation in /home/runner/work/wb-toolbox/wb-toolbox/build/doc/doxygen
2021-07-08T06:47:33.4654505Z [34mDEBUG    -  [0mLoading configuration file: /home/runner/work/wb-toolbox/wb-toolbox/doc/mkdocs/mkdocs.yml
2021-07-08T06:47:33.5660329Z [34mDEBUG    -  [0mLoaded theme configuration for 'material' from '/opt/hostedtoolcache/Python/3.8.10/x64/lib/python3.8/site-packages/material/mkdocs_theme.yml': {'language': 'en', 'direction': None, 'features': [], 'palette': {'primary': None, 'accent': None}, 'font': {'text': 'Roboto', 'code': 'Roboto Mono'}, 'icon': None, 'favicon': 'assets/images/favicon.png', 'include_search_page': False, 'search_index_only': True, 'static_templates': ['404.html']}
2021-07-08T06:47:33.5665581Z [33mWARNING  -  [0mConfig value: 'site_url'. Warning: This option is now required. Set to a valid URL or an empty string to avoid an error in a future release.
2021-07-08T06:47:33.5667900Z [33mWARNING  -  [0mConfig value: 'site_url'. Warning: The 'use_directory_urls' option has been disabled because 'site_url' contains an empty value. Either define a valid URL for 'site_url' or set 'use_directory_urls' to False.
2021-07-08T06:47:33.5669683Z [34mDEBUG    -  [0mConfig value: 'config_file_path' = '/home/runner/work/wb-toolbox/wb-toolbox/doc/mkdocs/mkdocs.yml'
2021-07-08T06:47:33.5671430Z [34mDEBUG    -  [0mConfig value: 'site_name' = 'WBT'
2021-07-08T06:47:33.5672182Z Aborted with 2 Configuration Warnings in 'strict' mode!
2021-07-08T06:47:33.5675910Z [34mDEBUG    -  [0mConfig value: 'nav' = [{'Home': 'index.md'}, {'Install': 'install.md'}, {'Troubleshooting': 'troubleshooting.md'}, {'Documentation [master]': 'https://robotology.github.io/wb-toolbox/doxygen/'}, {'Documentation [devel]': 'https://robotology.github.io/wb-toolbox/devel/doxygen/'}, {'Tutorials': [{'Introduction': 'tutorial_introduction.md'}, {'Basic Simulink simulation': 'simulink_basic.md'}, {'Use model blocks': 'simulink_model_blocks.md'}, {'Simulink and Gazebo': 'simulink_gazebo.md'}]}, {'Migration': [{'From WBI-Toolbox': 'Migration_from_WBI-Toolbox_1.0.md'}, {'From WB-Toolbox 2.0': 'Migration_from_WB-Toolbox_2.0.md'}]}, {'About': 'about.md'}, {'License': 'license.md'}]
2021-07-08T06:47:33.5678997Z [34mDEBUG    -  [0mConfig value: 'pages' = None
2021-07-08T06:47:33.5679959Z [34mDEBUG    -  [0mConfig value: 'site_url' = ''
2021-07-08T06:47:33.5681104Z [34mDEBUG    -  [0mConfig value: 'site_description' = 'A Simulink Toolbox for Whole-Body Control'
2021-07-08T06:47:33.5682318Z [34mDEBUG    -  [0mConfig value: 'site_author' = 'Diego Ferigo'
2021-07-08T06:47:33.5685589Z [34mDEBUG    -  [0mConfig value: 'theme' = Theme(name='material', dirs=['/opt/hostedtoolcache/Python/3.8.10/x64/lib/python3.8/site-packages/material', '/opt/hostedtoolcache/Python/3.8.10/x64/lib/python3.8/site-packages/mkdocs/templates'], static_templates=['404.html', 'sitemap.xml'], locale=Locale(language='en', territory=''), language='en', direction=None, features=[], palette={'primary': 'blue grey', 'accent': 'amber'}, font={'text': 'Roboto', 'code': 'Roboto Mono'}, icon={'logo': 'fontawesome/solid/cube'}, favicon='assets/images/favicon.png', include_search_page=False, search_index_only=True, feature={'tabs': True})
2021-07-08T06:47:33.5689145Z [34mDEBUG    -  [0mConfig value: 'docs_dir' = '/home/runner/work/wb-toolbox/wb-toolbox/doc/mkdocs/data'
2021-07-08T06:47:33.5690622Z [34mDEBUG    -  [0mConfig value: 'site_dir' = '/home/runner/work/wb-toolbox/wb-toolbox/build/doc/mkdocs'
2021-07-08T06:47:33.5692147Z [34mDEBUG    -  [0mConfig value: 'copyright' = 'Copyright &copy; 2020 Istituto Italiano di Tecnologia'
2021-07-08T06:47:33.5694195Z [34mDEBUG    -  [0mConfig value: 'google_analytics' = None
2021-07-08T06:47:33.5695525Z [34mDEBUG    -  [0mConfig value: 'dev_addr' = Address(host='127.0.0.1', port=8000)
2021-07-08T06:47:33.5696759Z [34mDEBUG    -  [0mConfig value: 'use_directory_urls' = False
2021-07-08T06:47:33.5698704Z [34mDEBUG    -  [0mConfig value: 'repo_url' = 'https://github.com/robotology/wb-toolbox/'
2021-07-08T06:47:33.5700196Z [34mDEBUG    -  [0mConfig value: 'repo_name' = 'robotology/wb-toolbox'
2021-07-08T06:47:33.5701912Z [34mDEBUG    -  [0mConfig value: 'edit_uri' = 'edit/master/doc/mkdocs/data/'
2021-07-08T06:47:33.5703335Z [34mDEBUG    -  [0mConfig value: 'extra_css' = []
2021-07-08T06:47:33.5704210Z [34mDEBUG    -  [0mConfig value: 'extra_javascript' = []
2021-07-08T06:47:33.5705529Z [34mDEBUG    -  [0mConfig value: 'extra_templates' = []
2021-07-08T06:47:33.5709695Z [34mDEBUG    -  [0mConfig value: 'markdown_extensions' = ['toc', 'tables', 'fenced_code', 'markdown.extensions.admonition', 'pymdownx.arithmatex', 'pymdownx.betterem', 'pymdownx.caret', 'pymdownx.critic', 'pymdownx.details', 'pymdownx.inlinehilite', 'pymdownx.highlight', 'pymdownx.keys', 'pymdownx.magiclink', 'pymdownx.mark', 'pymdownx.smartsymbols', 'pymdownx.superfences', 'pymdownx.tasklist', 'pymdownx.tilde', 'markdown [... cut ...]
2021-07-08T06:47:33.5927796Z make[3]: *** [doc/mkdocs/CMakeFiles/mkdocs.dir/build.make:71: doc/mkdocs/CMakeFiles/mkdocs] Error 1
2021-07-08T06:47:33.5931246Z make[2]: *** [CMakeFiles/Makefile2:157: doc/mkdocs/CMakeFiles/mkdocs.dir/all] Error 2
2021-07-08T06:47:33.5932726Z make[1]: *** [CMakeFiles/Makefile2:164: doc/mkdocs/CMakeFiles/mkdocs.dir/rule] Error 2
2021-07-08T06:47:33.5937024Z make: *** [Makefile:182: mkdocs] Error 2
~~~

See https://github.com/robotology/wb-toolbox/pull/222/checks?check_run_id=3016328818 .

diegoferigo, 2021-07-08:
Thanks for the ping. I assigned the issue to myself but I still didn't start debugging the problem.

traversaro, 2021-07-08:
Just to clarify, I posted in the issue because now the job is failing at the CI level (generating a red job), and not only generating a broken website (as before).

diegoferigo, 2021-07-13:
I fixed the generation in #223 but this problem persists. Locally the website renders correctly, but when deployed it does not.

I noticed that most of the documentation is now in blockfactory and the website mainly contains the installation instructions, what do you think @traversaro if we move the content to the README? There was originally the idea to create also some tutorials and for them the website would have been a better choice, but I doubt anyone will do that anytime soon.

traversaro, 2021-07-13:
Moving to the README is perfectly fine for me.

traversaro, 2021-11-26:
> Locally the website renders correctly, but when deployed it does not.

I quickly checked the difference, and while locally the website is rendered by the latest mkdocs, i.e. the source code of the website contained: 
~~~
<meta name="generator" content="mkdocs-1.2.3, mkdocs-material-7.3.6">
~~~
the deployed version contained:
~~~
<meta name="generator" content="mkdocs-1.1.2, mkdocs-material-6.0.2">
~~~

Then there are many more differences, but it is quite difficult to investigate more differences until this one is sorted out.

traversaro, 2021-11-26:
It seems that pip in the Actions is installing `mkdocs 1.2.3`, so something else is going on:
~~~
2021-11-26T09:52:51.5952636Z Collecting mkdocs
2021-11-26T09:52:51.6365902Z   Downloading mkdocs-1.2.3-py3-none-any.whl (6.4 MB)
~~~

traversaro, 2021-11-26:
I found part of the problem. The rest of the files get deployed correctly, but any file called `index.html` does not get copied due to this line (I guess, I did not checked): https://github.com/robotology/wb-toolbox/blob/f436bcf9742f1f12cbd11940acecd70e898e071b/.github/workflows/website.yml#L96 . To understand this, just check the time in which the files were last modified in the `gh-pages` branch . Any file not called `index.html` as the correct line:
~~~
      <meta name="generator" content="mkdocs-1.2.3, mkdocs-material-7.3.6">
~~~ 

However, this does not solves completly the problem as also the 404 page is not working correctly even if it corresponds to the locally generated one, see https://robotology.github.io/wb-toolbox/mkdocs/404.html .

traversaro, 2021-11-26:
No, it turns out that that was it, I fixed it locally and it works: https://traversaro.github.io/wb-toolbox/mkdocs/index.html .

closed 2021-11-26 by traversaro