robotology/wb-toolbox #210: QUESTION: what is the minimum MATLAB/Simulink version WBToolbox should support in the DIC organization?

nunoguedelha, 2021-05-18:
What is the minimum MATLAB/Simulink version WBToolbox should support in the DIC organization?

Actually I asked a more general question in Teams General, about the supported MATLAB version w.r.t. all the lab tools.

Ping @traversaro @diegoferigo

traversaro, 2021-05-19:
As a general line, I think that supporting more and less at least the last four version of MATLAB (so two years of releases) could make sense, to balance the maintenance effort with the possibility of dissemination (not everyone outside the lab may have the latest MATLAB version). However, this is entirely in theory as we don't actively test with older MATLAB versions and we do not have any MATLAB-based CI. For this reason, I think we can just evaluate on a case-by-case when incompatibilities arise (or when someone wants to use a feature only available in recent MATLAB versions) what to do.

diegoferigo, 2021-05-19:
In the past, we made an internal poll to check what was the oldest MATLAB installation, and took into account few downstream users outside IIT. Now, maybe we can define an official policy like what suggested by @traversaro, clearly document it, and start enforcing it in our exported slx that is the library that is publicly distributed.

traversaro, 2021-05-30:
Given that now MATLAB is available in GitHub Actions, I would try to add at least a smoke test for a block in the library, and then consider supported only the version that we can test in CI. From https://github.com/matlab-actions/overview#set-up-matlab, it seems that we can only test MATLAB >= 2020a, so unless there are specific requirements, I think we can consider supporting only MATLAB >= 2020a.

diegoferigo, 2021-05-31:
Totally agree to support >= 2020a. New decade-proof :)