robotology/wb-toolbox #215: Motor encoder values are improperly converted

S-Dafarra, 2021-05-24:
- [X] I already updated to the latest version I can use
- [X] I already checked similar issues using the search box
- [X] I already checked the website for known troubleshooting
- [X] I already cleaned my environment (by removing the wb-toolbox install folder and installing it again)
- [X] I already excluded the possibility that system configuration files (e.g. yarp related) might be responsible
- [X] I am sure the environment variables are correct
- [X] I tried to reproduce the error in simulation (if it affects the real robot)

#### Description

When retrieving the motor encoder values, they are converted from degrees to radians, but the inputs are not in degrees. Instead, the measurements are simply motor ticks.

This is the line:
https://github.com/robotology/wb-toolbox/blob/d9b0633fb289b594170ae270502c7b5d8cbe5372/toolbox/library/src/GetMeasurement.cpp#L285

cc @pattacini

S-Dafarra, 2021-05-24:
After a further check, it seems that the measurements are indeed in degrees. So, I think we can close this, right @pattacini?

pattacini, 2021-05-24:
Here you still convert to radians. Is this still wanted?

S-Dafarra, 2021-05-24:
> Here you still convert to radians. Is this still wanted?

I think so. All the blocks in the toolbox expect the joints to be specified in radians. This is also done for the joint encoders, for example https://github.com/robotology/wb-toolbox/blob/d9b0633fb289b594170ae270502c7b5d8cbe5372/toolbox/library/src/GetMeasurement.cpp#L225-L235

pattacini, 2021-05-24:
Great then!
We can close it.

closed 2021-05-24 by S-Dafarra

traversaro, 2021-05-24:
If we want to avoid this misunderstanding in the future, perhaps we should clarify this aspects in the [` yarp::dev::IMotorEncoders`](http://www.yarp.it/git-master/classyarp_1_1dev_1_1IMotorEncoders.html#af1aaa4d8b5fc732eb95a9d03bd16a569) docs.