robotology/wb-toolbox #224: Variant of Jacobian and DotJNu in which the frame is an input and not a parameter

traversaro, 2021-07-16:
#### Summary

It would be nice to have a variant of the Jacobian and DotJNu blocks in which the frame is given as input and not as parameter.

#### Motivation

@VenusPasandi asked about this, so probably she can provide a bit more details.

traversaro, 2021-07-16:
My first idea is to add a JacobianWithFrameAsInput and DotJNuWithFrameAsInput block that duplicate the existing blocks but with frame passed as input, but perhaps there is some more intelligent strategy.

VenusPasandi, 2021-07-16:
Actually, I found a way to use the current blocks  `Jacobian` and `DotJNu` in a Simulink model where the frame name (which is the parameter of these blocks) changes during the simulation. So, I can get Jacobian and Jacobian derivative of different frames by using only one block. For this purpose,
1) I create a Simulink function having the Jacobian/DotJNu blocks. Lets call it `mySimFunc`.
2) In a matlab function block, I use `set_param` to set the `Frame name` of the Jacobian/DotJNu blocks, (Note that you must use `coder.extrinsic('set_param')` in the matlab function to be allowed to use `set_param` command),
3) In the matlab function block, after setting the Frame name, I call `mySimFunc` which output is the Jacobian and Jacobian derivative of the desired frame set in the step 2.

VenusPasandi, 2021-07-16:
> My first idea is to add a JacobianWithFrameAsInput and DotJNuWithFrameAsInput block that duplicate the existing blocks but with frame passed as input, but perhaps there is some more intelligent strategy.

I think Simulink blocks do not accept a string as input signal. So, the block can take the `frame index` as the input.

traversaro, 2021-07-16:
> I think Simulink blocks do not accept a string as input signal. So, the block can take the `frame index` as the input.

Yes, that is correct for BlockFactory blocks, but just FYI in general Simulink blocks support string as signal type, see https://www.mathworks.com/help/simulink/ug/simulink-strings.html .

traversaro, 2021-07-16:
Given @VenusPasandi solution, I think we can close the issue.

closed 2021-07-16 by traversaro