robotology/wb-toolbox #225: Add relative Jacobian block

HosameldinMohamed, 2021-08-12:
#### Summary

A block similar to the `Jacobian` block,
[image]

but provided with 2 frame names as inputs, it gives the relative Jacobian between the 2 frames. 

It can implement the iDyntree method 

```cpp
bool KinDynComputations::getRelativeJacobian(const iDynTree::FrameIndex refFrameIndex,
                                             const iDynTree::FrameIndex frameIndex,
                                             iDynTree::MatrixDynSize & outJacobian)
```

In https://github.com/robotology/idyntree/blob/f9051752ca5a6ca8f3a4f101e4a536fe1eb62c5f/src/high-level/src/KinDynComputations.cpp#L1661-L1663

#### Motivation

It seems like a clean way to have the relative jacobian since the iDyntree function is already available, we just had to wrap it.

#### Additional context

We drafted a quick block (with @Giulero's help), it looks like this

[image]

It was a quick draft to test it with the robot, but I think we can omit the input ${ }^{world} H_{base}$.

traversaro, 2021-08-21:
> It was a quick draft to test it with the robot, but I think we can omit the input ${ }^{world} H_{base}$.

Just to clarify, this happens only because in iDynTree (with default velocity representation settings) the relative jacobian between two links L and C is the $J\_{L,C}^{rel}  $ matrix such that:

$$
{}^{C[L]} \mathrm[v}\_{L,C} = J\_{L,C}^{rel} \dot{s}
$$

where $s \in \mathbb{R}^{n_{dofs}}$ is the vector of internal joint positions (by the way, this is unfortunatly not fully documented). If the relative jacobian assume some other representation of the relative velocity between two frames, for example one using the absolute/world frame, then a dependency on ${}^W H_B$ could appear.

traversaro, 2024-09-24:
Fixed by https://github.com/robotology/wb-toolbox/pull/226 .

closed 2024-09-24 by traversaro