robotology/wb-toolbox #231: Solve osqp-eigen warning

traversaro, 2022-02-03:
Solve osqp warning:
~~~
/__w/robotology-superbuild/robotology-superbuild/src/WBToolbox/toolbox/library/src/OSQP.cpp: In member function 'virtual bool wbt::block::OSQP::output(const blockfactory::core::BlockInformation*)':
/__w/robotology-superbuild/robotology-superbuild/src/WBToolbox/toolbox/library/src/OSQP.cpp:549:50: warning: 'bool OsqpEigen::Solver::solve()' is deprecated: Use solveProblem() instead. [-Wdeprecated-declarations]
     bool solveReturnVal = pImpl->sqSolver->solve();
                                                  ^
~~~

I think it was introduced by osqp-eigen 0.7.0 .

YYB666, 2022-03-30:
I also meet the same warning, have you solved it?

traversaro, 2022-03-30:
No. However, it is just a warning, it should not prevent you from using the library.