robotology/wb-toolbox #233: Expose adaptive_rho option in OSQP block

traversaro, 2022-10-14:
#### Summary

We would like to expose the adaptive_rho  option (and eventually other options in the OSQP block.

#### Motivation

By default, OSQP is not deterministic, but if you set adaptive_rho to false, you get a deterministic solver.

#### Additional context

This emerged in https://github.com/ami-iit/matlab-whole-body-simulator/pull/81#issuecomment-1274395409 .

traversaro, 2022-10-14:
See https://github.com/robotology/wb-toolbox/issues/202 for a similar issue.

traversaro, 2022-10-14:
See https://github.com/robotology/wb-toolbox/blob/master/toolbox/library/src/OSQP.cpp .

VenusPasandi, 2022-10-17:
Reading the list of settings for OSQP and their role in the OSQP, I think 3 variables are worth having in the OSQP block:

- **adaptive_rho**: this is a flag to enable/disable the adaptive procedure for rho. The use of the adaptation algorithm increases the complexity and processing time for each time step, though it reduces the total processing time for complex problems. As a trade-off, OSQP uses the adaptation algorithm only when it is required. The realization of the problem's complexity depends on the processing time that depends don't the machine/computer specifications. So, enabling this parameter could make OSQP indeterministic.
- **polish**: this is a flag to enable/disable the final polishing algorithm for increasing the solution accuracy. This feature is disabled by default.
- **max_iter**: this is a number defining the maximum allowable iteration for OSQP.