robotology/wb-toolbox #236: Deterioration of performances when Simulink runs with an open scope (with real robot).

Giulero, 2022-10-25:
- [x] I already updated to the latest version I can use
- [x] I already checked similar issues using the search box
- [x] I already checked the website for known troubleshooting
- [ ] I already cleaned my environment (by removing the wb-toolbox install folder and installing it again)
- [ ] I already excluded the possibility that system configuration files (e.g. yarp related) might be responsible
- [x] I am sure the environment variables are correct
- [ ] I tried to reproduce the error in simulation (if it affects the real robot)

#### Description

The issue occurred during a joint data collection on the real robot (iCub Scarlett). The data were collected using a Simulink model that sent joint position references to the robot using a `setPosition` (direct) block. 

When a scope is open, the model's performance seems to deteriorate. In particular, when using `setPosition` (in position direct) the **actuated** trajectory is jerky, while the desired one is smooth. Once the scope is closed, the actuation becomes smooth again. 
The issue occurs even if the scope is already open before running the Simulink model. If instead the scope is opened during the execution of the model, the robot freezes for a while and catches the desired trajectory with a sudden movement. Then continues with the choppy behaviour. 

cc @gabrielenava 

#### Platform

We spotted the bug in a data collection in the real robot (iCub Scarlet).

#### Reproducibility

We tried many times with the real robot and the issue arose every time. Still, we tested only with one machine (`icub29`)
 
I cannot reproduce the issue in simulation. I guess the synchronization with Gazebo prevents it.

#### Screenshots

#### The desired vs the measured joint values 

[image]

[image]

A zoom

[image]

The blue trajectory (the measured joint values) is jerky.

#### Additional information

- OS: Ubuntu 20.04 
- Version / Commit: [e.g. `v4.0`]
- Matlab Version: [e.g. `2018a`]

@gabrielenava can you help me with this? I do not remember the versions of software on icub29 :(

traversaro, 2022-10-25:
I think in general this is a known aspect of Simulink: it does not provide you with any (even soft) real-time guarantees. If you want to run a controller in real time, you should use automatic code generation using Simulink Coder (supported in WB-Toolbox since WB-Toolbox 4), or Simulink Real-Time. 

For more info, see:
* https://github.com/robotology/wb-toolbox/issues/160
* https://github.com/robotology/whole-body-controllers/issues/67

traversaro, 2022-10-25:
> I cannot reproduce the issue in simulation. I guess the synchronization with Gazebo prevents it.

I think you can reproduce it if you disable the Gazebo synchronized and use the Real-Time one.

gabrielenava, 2022-11-09:
>Matlab Version:
@gabrielenava can you help me with this? I do not remember the versions of software on icub29 :(

it is MATLAB 2021b