robotology/wb-toolbox #237: bug when parsing the mask of WBT Configuration block

LoreMoretti, 2023-05-23:
- [ ] I already updated to the latest version I can use
- [ ] I already checked similar issues using the search box
- [ ] I already checked the website for known troubleshooting
- [ ] I already cleaned my environment (by removing the wb-toolbox install folder and installing it again)
- [ ] I am sure the environment variables are correct

#### Description

It seems it is not possible to define the block parameters through the option `Configuration from: Mask`.

#### Platform

Simulation with Matlab/Simulink 2022b on Linux 22.04 

#### Reproducibility

Just use any model which contains the WBT Configuration block, and in the block mask from the drop-down menu choose `Mask` instead of `Workspace`, and then fill in all the edit boxes of the `Data` tab.

For example, let's assume that your object `WBToolbox.Configuration` is contained in the field of a structure like `configSimulation.WBTConfigRobotSim`, you will then have to fill the `Data` tab as follows

| edit box  | value|
| ------------- | ------------- |
| Robot Name  | configSimulation.WBTConfigRobotSim.RobotName  |
| Urdf File  | configSimulation.WBTConfigRobotSim.UrdfFile  |
| Controlled Joints  | configSimulation.WBTConfigRobotSim.ControlledJoints  |
| Control Boards Names  | configSimulation.WBTConfigRobotSim.ControlBoardsNames  |
| Local Name  | configSimulation.WBTConfigRobotSim.LocalName  |
| Gravity Vector  | configSimulation.WBTConfigRobotSim.GravityVector |

If you then run the Simulink model you should see an error ( as the one reported in **Additional context**).

If instead, the option from `Workspace`, passing for example `configSimulation.WBTConfigRobotSim` (which stores the `WBToolbox.Configuration` object), the simulation runs just fine.

#### Screenshots
[image]

[image]

#### Additional information

- OS: [e.g. Ubuntu 22.04]
- Matlab Version: [e.g. `2022b`]

#### Additional context

Here the error message:

```
Error reported by S-function 'BlockFactory' in '[simulation_with_PID_control/Simulator/EstimateContactForcesKynematicsForwardDynamics/RobotDynWithContacts_ClosedChain/simulinkFunction_dyn_worldTransform_inContactFrames/First ForwardKinematics/S-Function](matlab:open_and_hilite_hyperlink ('simulation_with_PID_control/Simulator/EstimateContactForcesKynematicsForwardDynamics/RobotDynWithContacts_ClosedChain/simulinkFunction_dyn_worldTransform_inContactFrames/First ForwardKinematics/S-Function','error'))':
ResourceFinder couldn't find urdf file configSimulation.WBTConfigRobotSim.UrdfFile.
Failed to initialize the KinDynComputations object.
Cannot retrieve handle to KinDynComputations.
```

LoreMoretti, 2023-05-23:
cc @traversaro @fabiodinatale

traversaro, 2023-05-23:
Ah, now the issue is clear to me. What is happening is that the mask is taking `configSimulation.WBTConfigRobotSim.UrdfFile` as a string, instead of evaluating its value.

fabiodinatale, 2023-05-23:
> What is happening is that the mask is taking configSimulation.WBTConfigRobotSim.UrdfFile as a string, instead of evaluating its value.

Yes, exactly, and by testing one by one, this is the only parameter affected by this behavior.

traversaro, 2023-05-23:
Thanks, that is interesting. Indeed, if I recall correctly we saw this with someone at somepoint, but I forgot about this.

traversaro, 2023-05-24:
Today I did more debugging on the original model in which @lrapetti had the problem, and I could not find the root of the problem. It seems however that the value was returned not evaluated directly by MATLAB, as I was debugging directly the value returned by `ssGetSFcnParam` at https://github.com/robotology/blockfactory/blob/v0.8.3/sources/Simulink/src/SimulinkBlockInformationImpl.cpp#L430, and already there the values were not evaluated. The strange thing is that if I set a name of the variable that does not exists there, I get an error due to the fact that the variable does not exists! 

At this point, the next step is to reproduce the problem in a less complex case, to debug more easily.

xela-95, 2023-07-10:
Hi, 
I was building a library block exploiting some of the wb-toolbox blocks; I created then a top-level mask where the user could set the parameters used in my library block, among which the parameters needed by the "config" of wb-toolbox.
For the "config" block I choose the "from mask" option and passed the parameters defined in the top-level mask, as in the figure below.
[image]

When I try to simulate the system I get the following error:
```
Error evaluating ['InitFcn'](matlab:Simulink.internal.OpenCallbackParamsDialog(['forward_dynamics/SoftContactsSimulator/Configuration'],'InitFcn');) callback of SubSystem block (mask) '[forward_dynamics/SoftContactsSimulator/Configuration](matlab:open_and_hilite_hyperlink ('forward_dynamics/SoftContactsSimulator/Configuration','error'))'. 
Callback string is 'source = get_param(gcb,'ConfigSource');

if strcmp(source, 'Workspace')
    object = get_param(gcb,'ConfigObject');
    WBToolbox.ConfigurationToMask(gcb, object);
end

WBToolbox.MaskToConfiguration(gcb);

clear object source'
Caused by:
Unrecognized function or variable 'wbt_controlled_joints'. 
Variable 'wbt_controlled_joints' does not exist. 
Suggested Actions
Load a file into base workspace.
Fix
Create a new variable.
```

CC @traversaro @LoreMoretti