robotology/wb-toolbox #241: Using OSQP solver in Simulink/Code generation/Compiling in Embedded MCU

EricWang0306, 2023-12-06:
Sorry for replying an old issue and I'm a rookie in git and github.
I will try my best to tell the details of my installation process and my project aim in text in case I miss some important information.

Project aim:
I just use the OSQP solver recently, and my purpose is to trans my simulink model with OSQP solver into C code and deploy the code onto our MCU, an Infineon 32-bit-tricore-microcontroller-TC397. 

Background:
Actually, we use the method of build a S-Function in simulink to calculate, and simuliate correct. And then the errors occur just for compiling the code generated from the model in simulink in Aurix Development Studio. And the errors show that some .h files lacks, but the .h files are Disorganized for their source. They are from like VC environment like windows.h and MATLAB source and so on. The total amount of these files is very large, and even if they are added, the code will not be recognized during compilation. So we think we can not use the way to compile the code and run the code in the MCU, including the OSQP solver. So we are finding the way to make the OSQP into a simulink inner function, so that our deployment to embedded controller may be proceed smoothly.

Process：
We build our control model in simulink and use OSQP as main solver in MPC algorithm. 
1.First, I downloaded the test file 'https://github.com/robotology/wb-toolbox/files/6528910/qpoases_vs_osqp_test_v3.zip'. Error shows there is no WB-Toolbox in the working path when I just run it directly. 
2.So I download the WB-Toolbox files(https://github.com/robotology/wb-toolbox/) and set them into 'D:\Program Files\Polyspace\R2019b\toolbox'. Add the path to the working path and WB-Toolbox can be found in simulink library browser. I run the test model but there are still other errors. First is 'error in S-Function 'qpoases_vs_osqp_test_v3/OSQP' S-Function ‘BlockFactory‘ does not exist.' QP module has the same error. Second is when I try to adjust parameters in OSQP module it will shows three question marks on the OSQP block. I don't know whether it is an error because I haven't run the test model with no error. 
3.Then I find the BlockFactory code(https://github.com/robotology/blockfactory/tree/master) and do the same things I just did to WB-Toolbox ifiles. But it didn't work. The errors still there.

Risk:
1. I've noticed that t requires OsqpEigen to be installed and the WBT_USES_OSQPEIGEN option to be set. But I've not succeed in installing it yet.

Ask for Help:
1. Is there any way to solve the problem in process above? 
2. Whether this way for using OSQP can generate code and compile without linking to any external library.

DanielePucci, 2023-12-06:
CC @diegoferigo

traversaro, 2023-12-07:
>  I just use the OSQP solver recently, and my purpose is to trans my simulink model with OSQP solver into C code and deploy the code onto our MCU, an Infineon 32-bit-tricore-microcontroller-TC397.

Unless your firmware is able to dlopen library, this is not possible, as blockfactory uses dlopen (on *nix) or LoadLibrary to open the library corresponding to each library of blocks, see https://github.com/robotology/blockfactory/blob/b4d198b09aa27822ae9ea6f97387164416c36e33/deps/sharedlibpp/src/SharedLibrary.cpp#L71 .

> 2.So I download the WB-Toolbox files(https://github.com/robotology/wb-toolbox/) and set them into 'D:\Program Files\Polyspace\R2019b\toolbox'. Add the path to the working path and WB-Toolbox can be found in simulink library browser

`blockfactory` and `wb-toolbox`  are compiled library, so you can't just download their source code and add them to the path. You need either to install them in binary form (as documented for example in https://github.com/robotology/robotology-superbuild/blob/master/doc/matlab-one-line-install.md) or compiling them from source using a C++ compiler, as documented for example in https://github.com/robotology/robotology-superbuild/ .