robotology/wb-toolbox #251: Errors partially/incorrectly displayed in matlab R2024b

AntonioViscomi, 2025-02-03:
- [ x ] I already updated to the latest version I can use
- [ x ] I already checked similar issues using the search box
- [ x ] I already checked the website for known troubleshooting
- [ ] I already cleaned my environment (by removing the wb-toolbox install folder and installing it again)
- [ x ] I already excluded the possibility that system configuration files (e.g. yarp related) might be responsible
- [ ] I am sure the environment variables are correct
- [ ] I tried to reproduce the error in simulation (if it affects the real robot)

#### Description

It is not possible to determine the exact source of the error, as the error messages are only partially or incorrectly displayed.

#### Platform

The bug occurs while running tests on the Camozzi setup ([config files](https://github.com/robotology/robots-configuration/tree/devel/experimentalSetups/ankleSetup)).

#### Reproducibility

#### Screenshots
Simulink terminal output:
[image]

| Main schematic 	| GetPositionMeasurement 	| YARP_vs_Simulink_execution_step 	|
|----------------	|------------------------	|---------------------------------	|
| [image]| [image]| [image]|

#### Additional information

- OS: Windows 11 (source installation via pixi) 
- Version / Commit: 7804a8e5a03b04e8eb3ee5589b941960e76ba298
- Matlab Version: R2024b

#### Additional context
I have checked the source code ([here](https://github.com/robotology/wb-toolbox/blob/master/toolbox/library/src/SetReferences.cpp)), but there are multiple errors that could be associated with the line containing "failed t".

The WBtoolbox is configured via matlab:
```matlab
%cleanup
close all
clear, clc
%% Robot configuration for WBToolbox
% TO DO: Execute at simulink startup
WBTConfigRobot           = WBToolbox.Configuration;
WBTConfigRobot.RobotName = 'ankleSetup';
WBTConfigRobot.UrdfFile  = 'model.urdf';
WBTConfigRobot.LocalName = 'WBT';

% Controlboards and joints list. Each joint is associated to the corresponding controlboard 
WBTConfigRobot.ControlBoardsNames     = {'ankle_setup_control_board'};
WBTConfigRobot.ControlledJoints       = {'ankle_joint'};

ControlBoards                                        = struct();
ControlBoards.(WBTConfigRobot.ControlBoardsNames{1}) = {'ankle_joint'};
```

traversaro, 2025-02-03:
Just to understand, so the problem is that `bfError` is not displayed in the MATLAB/Simulink terminal output, right? 

Did you checked if there is anything relevant in the terminal where you launched MATLAB?

AntonioViscomi, 2025-02-07:
> Did you checked if there is anything relevant in the terminal where you launched MATLAB?

I cannot find anything because when I use pixi to start matlab, the environment is exited as soon as matlab is started.
After some experimenting I can say that the errors are displayed because the yarprobotinterface is shutting down and consequently Simulink is shutting down.